toblum/McLighting #309: Hangup with many pixels

codmpm, 2019-01-08:
Hey,

I've recently installed some new McLightning strips in the same bar I already had a bigger installation running (https://allgeek.de/2018/01/29/esp8266-neopixel-controller/ - German).

The new ones have 370 and 740 RGB (WS2812) pixels attached and I am using a more fitting LDO on the boards mentioned in the blog.
Now, after something like 2-3 hours, the larger strip does not work anymore - a simple reset and everything is fine again for the next hours. After that discovery the owner of the bar told me, that the old installation behaves in a similar way.
The 370 pixel strip is working all the time without any issues.

Is there any posibility, that depending on the number of pixels, there's something like a memory-leak? Do I have any option to test/reproduce this?

Cheers,
Patrik

ryansabin, 2019-01-08:
I found this same issue with a large installation that I have running. The problem ended up being voltage. Using a 5v 30amp power supply I set the voltage right at 5v. When I changed the voltage down to 4.95v everything started working fine.

Not sure why this change helped me. I believe that when the lights go from on to off or start using lots of power that the voltage fluctuates causing the power supply to compensate and sometimes overpower the board. This caused a reboot of the board or even worse it quit working until power cycle.

Try this and let me know your experience. I saw this with 2 out of my 4 x 500 led installations using Wemos D1 minis. I will be upgrading the controllers to NodeMCU boards next week.

codmpm, 2019-01-08:
We have 3 x 10A MeanWell power supplies that feed the stripe every 4-5m.

I think you have a very good point and I will check this as soon as possible. Thank you!!
I'm using my self-designed pcb's and had already experimented with different LDO's. Currently I'm using the [AP2112](https://www.diodes.com/assets/Datasheets/AP2112.pdf) with which I had more stable results as with the [AP2210](https://www.diodes.com/assets/Datasheets/AP2210.pdf).
The Wemos D1 is using the [ME6211](http://www.jytone.com.cn/upfile/file/20150404/2015040416594363694.pdf).

Are the LDO's able to compensate the behavior that you described?

ryansabin, 2019-01-08:
Not sure on your question. I can also point out that if you over power the Wemos D1 mini you will have many weird things happen including the reboots and hangs. We had 1 board that would speed up everything. Others would light random pixels random colors. Definitely make sure you are not going over 5v.

codmpm, 2019-01-08:
I was thinking on how the ESP will get unstable if the LDO is the "thing" that delivers the 3.3V for it.

Nevertheless, I will test to go to 4.9V for the stripes and therefore for the input of the LDO for the ESP.

debsahu, 2019-01-08:
I'm guessing there is no serial output available?

DennisH1978, 2019-02-09:
Hello, same Issue here. I got 1300 LED's - every 2m feeded from a 80A supply. 4,95V input yet tried.
But the programs will not work in the speed how my other 2 stripes (2m) are working. Let's say programm 14 runs in the little stripes very fast. In the big one with the 1300 LED's the it runs very slow. What can i do?

debsahu, 2019-02-09:
@DennisH1978 speed is relative to the number of pixels. We use a non linear scaling function to map 0-256 to 0-65k https://github.com/toblum/McLighting/blob/1578cafbf0aee2cbec9e5e1b76be3a6754072837/Arduino/McLighting/request_handlers.h#L96 feel free to change this line to your liking

codmpm, 2019-02-09:
@DennisH1978, I've wrote about the speed difference a while ago. https://allgeek.de/2018/01/29/esp8266-neopixel-controller/ (german)

DennisH1978, 2019-02-10:
@debsahu thx, but i've changed every number just so see, what happens. Nothing. What did i do wrong?
There is no difference. Could you tell me, which number is for what function, so i understand this?
@ codmpm thx fpr your link, but if i got my success without building an whole new controller, i try this first :-)

debsahu, 2019-02-10:
@DennisH1978 I used MATLAB to simulate non-linearity and optimized these values for upto 300 LEDs. Can't do this math off the top of my head and you can come up with any scaling function to map these values. I used a simple exponential function with offset.

Since this is off-topic I suggest you open a separate issue and we can come up with a solution.

codmpm, 2019-02-10:
> @ codmpm thx fpr your link, but if i got my success without building an whole new controller, i try this first :-)

You're welcome, but that's not what I meant. I explain in the article, that the speed differences are coming from the different pixelcount.

debsahu, 2019-02-21:
@codmpm Are you using ESP8266-Core v2.4.2? Others in WLED project have reported issues with 300 or more LEDs under DMA method. Looks like new core is a bit RAM heavy. Which PIN are you using? Could you switch to UART mode?

codmpm, 2019-02-21:
Hey... pretty sure I'm using Core 2.3.x... and currently GPIO as this is the "old" design of my board.