toblum/McLighting #312: Web server dies after time - and the light still goes on

kdupke, 2019-01-09:
Thank you for this great piece of software.

Unfortunately I have an issue with the web interface.

After a time of inactivity (66 seconds) I can't reach it any more.

If I started an effect as last action, the effect will still go on.

There is no error message on the serial console.

This is an 1M ESP-01S.

Any idea how to debug/dig into this?

esp_status when working:
HOSTNAME | "McLighting01"
version | "2.1.11"
heap | 27496
sketch_size | 400368
free_sketch_space | 495616
flash_chip_size | 1048576
flash_chip_real_size | 1048576
flash_chip_speed | 40000000
sdk_version | "2.2.1(cfd48f3)"
core_version | "2_4_2"
cpu_freq | 80
chip_id | 1335429
animation_lib | "WS2812FX"
pin | 2
number_leds | 12
button_mode | "ON"
home_assistant | "OFF"
legacy_animations | "ON"
state_save | "SPIFFS"

mike2nl, 2019-01-17:
@kdupke 
Hi,
i'm Mike from Tasmota an alternaive firmware for the sonoff devices from iTead.cc.

I use much NEOPixel animation in them and was searching for some info here for McL.
I had read your issue and you use an old core: core_version | "2_4_2"

Switch to the core 2.5.0-beta2 in Arduino IDE. Update it via the board manager.

...because there are many things changed and the dying web server is known as
an issue in core 2.4.2. NOT on all systems it's depending on many factors.
Network stability, Router/AP type, much used WIFI channels means to many other users,
how the AP is placed in the house, and so on.

Test the new core 2.5.0-beta2. 
For tasmota we have changed too because the new core solves a lot of issues. 
I hope it helps you

toblum, 2019-01-20:
Hi @kdupke,

I think @mike2nl is right. There were issues with this core version. I think that should solve your problem. Have a look at the detailed settings listed here: https://github.com/toblum/McLighting/wiki/Software-installation

@mike2nl Great to have you here as I'm a big fan of Tasmota. I like it for the great support/documentation and the tons of features.

Regards
Tobias

kdupke, 2019-01-22:
I now switched to 2.5.0-beta2, unfortunately behavior has not changed.

What I mentioned, this happens with both the wifi config manager as well as the configured mclighting.

sdk_version | "3.0.0-dev(c0f7b44)"
core_version | "2_5_0_BETA2"

Any idea what to do next?

best regards,
kai

debsahu, 2019-01-24:
Try a different hardware, looks like something wrong with the ESP8266. Do you have a 100uF (1000uF is even better) near your ESP8266 between 3.3V and GND? Also check your power supply to see if are seeing a brownout.

kdupke, 2019-01-25:
OK, I tested now with 3 different ESP-01s/ESP8266 (from different sources), with a 5amps power supply - without any LED stripes.

(I do not have other esp8266 available to test, only ESP32 and ESP-01s)

I also tried with a 470µF I had handy.

The web server dies as another day.

Just to be clear, it works like a charm, when I do not let the web access idle for 66 seconds or more. When I do every 65 or less seconds a wget on the status, then the application keeps working like a charm.

Any other ideas?

Best regards,
kai

kdupke, 2019-02-01:
I tried, and it does not change the behavior.

It also disables sleep mode, right?

Is there any what happens, say, after 60s somewhere? An internal timer or so?

debsahu, 2019-02-01:
Nothing is hardcoded in the code that would account for the 60s. https://github.com/toblum/McLighting/blob/1578cafbf0aee2cbec9e5e1b76be3a6754072837/Arduino/McLighting/McLighting.ino#L957 is run every loop.
I'm wondering this has something to do with your router renewing the lease very 60s? << Hitting in the dark.
Just for giggles, could you try uploaded https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/examples/HelloServer/HelloServer.ino and see if you able to replicate this issue?

kdupke, 2019-02-01:
Renew for IP addresses is every 2h.
HelloServer dies as well, with and without WiFi.setSleepMode(WIFI_NONE_SLEEP);

Enabling debug last line is pm open, type:0 0 - nothing afterwards, just silent stopping working

So, it does not seems to be the an issue of McLighting. Which is good for this project, but does not help me of course....

debsahu, 2019-02-05:
Closing as this issue is not related to McLighting. Wish I knew more about routers to help you out.

closed 2019-02-05 by debsahu

kdupke, 2019-02-08:
@debsahu this isn't router related, as it already happens in ad-hoc mode when wifi setup is running.

I agree this seems to be related to the hardware, so not an issue of McLighting.

debsahu, 2019-02-08:
The in-built webserver is riddled with issues. @darkain worked on rewriting this completely and was temporarily abandoned because of underlying issues with WiFi stack. Now there is an attempt to fix this here https://github.com/esp8266/Arduino/pull/5636