toblum/McLighting #313: Nach Hochladen des Codes kein Access Piont

Oemer9119, 2019-01-11:
Hallo,

Habe mir Leds bestellt, aber bis dahin wollte ich den Code etwas ausprobieren.
Leider klappt das nicht so immer. Nach dem Hochladen sollte ich mich mit dem Wlan vom ESP8266 verbinden um ihn zu konfigurieren, aber da erscheint keiner. Beim ersten mal hat es geklappt, danach lange nicht mehr dann wieder einmal und jetzt kommt nichts mehr. 
Vielen dank für deine mühe echt tooler Code.

M4lik, 2019-01-11:
Das Problem an der Stelle ist, dass beim flashen des ESP nur der Teil des Speichers geschrieben wird, der auch von der Firmware in anspruch genommen wird. Dadurch dass die WLAN Einstellungen aber weiter hinten im Speicher liegen werden diese nicht überschrieben.

Ich würde Empfehlen mithilfe eines Tools (wenn du auf einem *NIX System unterwegs bist ist [`esptool`](https://github.com/espressif/esptool) zu empfehlen) den gesammten flash zu nullen und dann die Firmware neu aufzuspielen.

Oemer9119, 2019-01-11:
Danke es Funktioniert nach dem ich die einstellung gemacht habe.
[image]

debsahu, 2019-01-11:
close?

closed 2019-01-11 by Oemer9119