toblum/McLighting #318: I am Stuck

JPluess, 2019-01-24:
hello guys, i have it installed and WIFI/web interface works, i have set the header file to;

// Neopixel
#define PIN 5           // PIN (14 / D5) where neopixel / WS2811 strip is attached
#define NUMLEDS 300       // Number of leds in the strip
#define BUILTIN_LED 2    // ESP-12F has the built in LED on GPIO2, see https://github.com/esp8266/Arduino/issues/2192
#define BUTTON 3         // Input pin (4 / D2) for switching the LED strip on / off, connect this PIN to ground to trigger button.
#define SKETCH_VERSION 1 // needs to be added if you use Visual Micro

i have checked with the oscilloscope there are no pulses on output d5, changing it to any other wont work either. other libraries have worked with d9. 

could anybody help me?

debsahu, 2019-01-24:
5 means GPIO5. If you want to use D5 replace 5 with D5.

JPluess, 2019-01-24:
hi debashu, i understand, yet it won’t work, i get some random output at d5, at random intervals.

JPluess, 2019-01-24:
i have reinstalled ws2812fx library, now it works. thanks alot.

version | "2.1.11"
-- | --
heap | 19704
sketch_size | 588784
free_sketch_space | 2555904
flash_chip_size | 4194304
flash_chip_real_size | 4194304
flash_chip_speed | 40000000
sdk_version | "3.0.0-dev(c0f7b44)"
core_version | "2_5_0_BETA2"
cpu_freq | 80
chip_id | 1458400
animation_lib | "WS2812FX"
pin | 2
number_leds | 300
button_mode | "ON"
amqtt | "ON"
home_assistant | "ON"
legacy_animations | "ON"
esp8266_http_updateserver | "ON"
state_save | "SPIFFS"

closed 2019-01-24 by JPluess