toblum/McLighting #319: file management

JPluess, 2019-01-24:
i just run into an issue having a probably interrupted index2.htm upload transfer, resulting in a broken file.

Behavior on startup when serial monitor is on it froze at the time it opened the spiffs readout. i could start just fine without the serial monitor on. requesting over html the index2.htm would result in a crash. I fixed it with a simple sketch to format spiffs.

my idea1: if you add a simple button to the http://McLighting-ip/edit for formatting spiffs or
my idea2: if you want to make a checkbox to delete files individually. 

Idea3: if no files are uploaded yet, why not make http://mclighting-ip/upload as default address?

Thanks for consideration

debsahu, 2019-01-24:
I have a PROGMEM version in [develop branch](https://github.com/toblum/McLighting/tree/develop). No uploading required, uncomment `#define USE_HTML_MIN_GZ` in **definitions.h**. Test it out and let us know.

debsahu, 2019-01-25:
Just added E131 mode, turn the E131 mode and send data from jinx

JPluess, 2019-01-25:
great how can i download the latest update?
i have 1 strip 300 led running just fine with the library https://github.com/rstephan/ArtnetWifi
setup in Jinx is RGB style, 2 Universes
by searching i found a sketch that can record to sd card and playback. 
https://github.com/tangophi/Artnet_DMX_SD_Card
but why not use spiffs as location?

JPluess, 2019-01-25:
oh i see, i understood just as just today. ;)
is there any chance for saving to spiffs?

debsahu, 2019-01-25:
Not practical!

> is there any chance for saving to spiffs?

3(r,g,b) * 300(no_of_pixels) * 30(fps) = 26.4kB is the amount of space you need for 1 second of data. With 3.5MB of SPIFFs, you can store a max of 97 seconds of data. Writing code for capturing and retransmitting is not an easy task or a priority. If you are able to write those code and is lightweight enough that it can do the transmitting and receiving at the same time, please do a PR.

A more practical approach would be store rgb data as BMP in the SPIFFs and read it sequentially (as avertical strip). I have some code written using [this approach here](https://github.com/debsahu/ESP8266-NeoAirPainter), but that is not relevant to McLighting project.

closed 2019-01-28 by debsahu