toblum/McLighting #325: Request: Split LEDs in several Areas

FraGThieF, 2019-02-01:
Hi,
Thank you so much for this project.
Because of this project, i startet with arduino!

Is it possible to ad areas in the code?

In my case it would be something like 300 LEDs in total with one dataline, splitet in areas like

Top         = 1-x
Bottom   = x+1 - y
Cirle 1     = y+1 - z
Square     = z+1 - 300

I would like to switch them seperate on and of.

debsahu, 2019-02-01:
There are multiple ways to approach this,
1. Use E1.31 to setup and use the E131 mode. Setup your installation on xlights. Example instructions can be found [in video here](https://www.youtube.com/watch?v=lZ09GlO2_8s)
2. Use segments in WS2812FX [example](https://github.com/kitesurfer1404/WS2812FX/blob/master/examples/ws2812fx_segments/ws2812fx_segments.ino)
3. Use websockets API [wiki](https://github.com/toblum/McLighting/wiki/WebSocket-API#payload-numledhexrgbnumledhexrgbnumledhexrgb-)

FraGThieF, 2019-02-05:
> There are multiple ways to approach this,
>
> 1. Use E1.31 to setup and use the E131 mode. Setup your installation on xlights. Example instructions can be found [in video [...]

Thank you for your help.

I guess i choose number 2, because they are all in still in 1 dataline.

But i struggle a little bit, where i have to add this code.

Thank you in advance for yur help

debsahu, 2019-02-05:
"2" means you abandon McLighting and write your own code / animation. It can be integrated to McLighting, but not the first thing on the todo of this project. But adding segments is the most requested feature here, needs significant modifications in WebUI.

Mezirx, 2019-04-08:
Is this feature already built in?

debsahu, 2019-04-09:
1 and 3 are

closed 2019-04-09 by debsahu