toblum/McLighting #326: Simpler MQTT/Home Assistant Integration

TECHNOFAB11, 2019-02-04:
Please add the possibility to configurate these more easily (best would be with the WiFi Manager/in the AP Config mode):
- Hostname
- MQTT Topic (would be better to be able to use "Home/Room/Device" for example) -> change "Home/Room", device will be Hostname

Would be awesome :D
Thanks for reading

debsahu, 2019-02-05:
Please edit this line https://github.com/toblum/McLighting/blob/1578cafbf0aee2cbec9e5e1b76be3a6754072837/Arduino/McLighting/definitions.h#L81 to your liking!

closed 2019-02-05 by debsahu

TECHNOFAB11, 2019-02-07:
Thanks, I did this already and changed some parts so they are a bit easier to change, but it would be cool to change it easier :D I know, changing lines is easy, but with Ui/ putting these Lines to the start of the file _definitions.h_ would be helpful for other people I guess. It's only a improvement, so it's not important.