toblum/McLighting #327: Upcoming Home Assistant version - Stricter MQTT config

aidbish, 2019-02-05:
In the next version of home assistant (0.87) stricter controls have been implemented.
Asa  result the MQTT discovery and manual config now have errors
The on_command_type is no longer supported for Json schema

please see this issue for further information

https://github.com/home-assistant/home-assistant/issues/20685

debsahu, 2019-02-05:
@aidbish Thanks! Why are they breaking things ?!?! 😣

Comment out `json["on_command_type"] = "first";` and check if it works?

What that line did was send ON command first from HA every time color change is done. The default was it sent color first and then ON command later. I need to test with MQTTfx or something to see what are the order of MQTT commands being sent.. 😵

aidbish, 2019-02-05:
I think their intention is to clean up old code and make consistent
commenting out the on_command_type in the manual config , the errors stops but have not been able to test the opeartion of the light yet

i will have a play with other discovered lights and modify the code and report back over next few days

rgogada, 2019-02-07:
yup, it happened. just updated to 0.87 and all my mclighting strips (12 of them) are now dead. could someone advice if I should go back to 0.86.4? thanks.

aidbish, 2019-02-07:
Have you manually defined the lights? if so remove on_command_type from the config.
If using discovery we need to wait for new version that is compatible

@debsahu Havn't forgot about testing, just busy with work at moment

rgogada, 2019-02-07:
I am using discovery so nothing to remove. However what worked for is the below.

1) comment out the two lines with 'on_command_type' (line #1022 & 1109) in request_handlers.h 
2) recompile and update thru OTA.

debsahu, 2019-02-07:
@aidbish no worries, same on my end as well. Have to test the develop version too on my end, see if proposed changes is acceptable to @toblum, add this change in and maybe work on 2nd/3rd color settings as proposed in another issue.

@rgogada thanks for the input, can you confirm that on/off, changing colors, animations and brightness works after uploading the changes?

rgogada, 2019-02-08:
@debsahu yes, everything works.

debsahu, 2019-03-05:
I check with 0.88.1 and everything seems to be working.  We will be merging with master soon.

closed 2019-03-07 by debsahu