toblum/McLighting #328: Question: Candle Simulation

ranseyer, 2019-02-07:
Hi, 
im thinking about building 6 pieces of this WiFi Candle: https://www.letscontrolit.com/forum/viewtopic.php?f=2&t=2147#p10092

But the code is very hard coded to 4*5 LEDs. Is there an easy Replacement possible with McLighning ?
Regards
Ranseyer

debsahu, 2019-02-13:
Animation you are looking for is Fire2012 which is FastLED based. See following [code](https://github.com/kitesurfer1404/WS2812FX/blob/master/examples/ws2812fx_custom_FastLED/ws2812fx_custom_FastLED.ino) Once can incorporate this in McLigthing, but stock Mclighting doesn't have this specific effect.

debsahu, 2019-02-13:
In **McLigthing.ino**,

1. Add FastLED header
```
#include <FastLED.h>
```
2. In setup(), add
```
strip.setMode(FX_MODE_CUSTOM);
strip.setCustomMode(myCustomEffect);
```
3. Somewhere before setup()
```
uint16_t myCustomEffect() {
  Fire2012();

  // return the animation speed based on the ws2812fx speed setting
  return (strip.getSpeed() / NUMLEDS);
}

/*
 * paste in the Fire2012 code with a small edit at the end which uses the
 * setPixelColor() function to copy the color data to the ws2812fx instance. 
*/

// Fire2012 by Mark Kriegsman, July 2012
// as part of "Five Elements" shown here: http://youtu.be/knWiGsmgycY
//// 
// This basic one-dimensional 'fire' simulation works roughly as follows:
// There's a underlying array of 'heat' cells, that model the temperature
// at each point along the line.  Every cycle through the simulation, 
// four steps are performed:
//  1) All cells cool down a little bit, losing heat to the air
//  2) The heat from each cell drifts 'up' and diffuses a little
//  3) Sometimes randomly new 'sparks' of heat are added at the bottom
//  4) The heat from each cell is rendered as a color into the leds array
//     The heat-to-color mapping uses a black-body radiation approximation.
//
// Temperature is in arbitrary units from 0 (cold black) to 255 (white hot).
//
// This simulation scales it self a bit depending on NUM_LEDS; it should look
// "OK" on anywhere from 20 to 100 LEDs without too much tweaking. 
//
// I recommend running this simulation at anywhere from 30-100 frames per second,
// meaning an interframe delay of about 10-35 milliseconds.
//
// Looks best on a high-density LED setup (60+ pixels/meter).
//
//
// There are two main parameters you can play with to control the look and
// feel of your fire: COOLING (used in step 1 above), and SPARKING (used
// in step 3 above).
//
// COOLING: How much does the air cool as it rises?
// Less cooling = taller flames.  More cooling = shorter flames.
// Default 50, suggested range 20-100 
#define COOLING  55

// SPARKING: What chance (out of 255) is there that a new spark will be lit?
// Higher chance = more roaring fire.  Lower chance = more flickery fire.
// Default 120, suggested range 50-200.
#define SPARKING 120

void Fire2012()
{
// Array of temperature readings at each simulation cell
  static byte heat[NUMLEDS];

  // Step 1.  Cool down every cell a little
    for( int i = 0; i < NUMLEDS; i++) {
      heat[i] = qsub8( heat[i],  random8(0, ((COOLING * 10) / NUMLEDS) + 2));
    }
  
    // Step 2.  Heat from each cell drifts 'up' and diffuses a little
    for( int k= NUMLEDS- 1; k >= 2; k--) {
      heat[k] = (heat[k - 1] + heat[k - 2] + heat[k - 2] ) / 3;
    }
    
    // Step 3.  Randomly ignite new 'sparks' of heat near the bottom
    if( random8() < SPARKING ) {
      int y = random8(7);
      heat[y] = qadd8( heat[y], random8(160,255) );
    }

    // Step 4.  Map from heat cells to LED colors
    for( int j = 0; j < NUMLEDS; j++) {
      CRGB color = HeatColor( heat[j]);
      int pixelnumber;
      if( gReverseDirection ) {
        pixelnumber = (NUMLEDS-1) - j;
      } else {
        pixelnumber = j;
      }

// **** modified for use with WS2812FX ****
//    leds[pixelnumber] = color;
      ws2812fx.setPixelColor(pixelnumber, color.red, color.green, color.blue);
// **** modified for use with WS2812FX ****

    }
}
```

debsahu, 2019-02-26:
Your question is answered, please reopen if you are still facing issue.

closed 2019-02-26 by debsahu