toblum/McLighting #329: Many LED's and less speed

DennisH1978, 2019-02-10:
Hi there. I got 1300 LED's - every 2m feeded from a 80A supply. 4,95 and 5V input yet tried.
But the programs will not work in the speed how my other 2 stripes (2m) are working. Let's say programm 14 runs in the little stripes very fast. In the big one with the 1300 LED's the it runs very slow. What can i do?
Member "debsahu" told me to change the numerbs in this:

uint16_t ws2812_speed = 61760 * (exp(0.0002336 * mcl_speed) - exp(-0.03181 * mcl_speed));

I did, but no change, don't matter what i am changing and in which direction.
Someone got an idea how to get the programs by 1300 LED's faster?

debsahu, 2019-02-10:
Which animation are you using? Let's optimize for that first.

DennisH1978, 2019-02-11:
Hi debsahu... The Problem is on all Animations i tested. Very good to See is on Animation 14, Dual scan.

DennisH1978, 2019-02-17:
Hi there... Do you need more Information about the setup or anything Else?

joeflower, 2019-02-21:
Take a look here. Perhaps a solution.
https://allgeek.de/2018/01/29/esp8266-neopixel-controller/

DennisH1978, 2019-02-21:
Hello joeflower... Thx, but that is not a solution I could use. But thank
you 😊

debsahu, 2019-02-21:
Hi @DennisH1978, I did some testing. As @codmpm and @joeflower have pointed out, no matter the highest speed the animations are slow for number of LEDs you have (1300). Unfortunately the animation is dependent on WS2812FX.

@kitesurfer1404 and @moose4lord is there a way to speed up the animations for large number of LEDs?

kitesurfer1404, 2019-02-21:
There might be some general code optimization possible. Or changing effects to assembly code and optimize them for large installations. But that's not the focus of the WS2812FX lib. It should provide good readable code and effect examples that can be understood by most programmers and modified easily if needed.
@DennisH1978 If not all FXs are needed, try to optimize them for your large installation. You might want to try to hardcode your parameters in and do some heavy loop unrolling.

DennisH1978, 2019-02-21:
Thx kitesurfer... I would love to Do this. But I am not very professional at programming something like this. If you could tell me, what parameters I have to change, I would try and error 😊 and if it works, o know what to Do with the other Animations 😀

moose4lord, 2019-02-21:
1300 LEDs is really stretching the capabilities of the ESP8266. It takes about 40ms just to write data to the LEDs, so, at best, you can only get 25 animation frames per second. And with so much time spent writing to the LEDs, the WiFi will probably be starved for processor time and you'd likely have an unreliable network connection. I think partitioning your LEDs and using separate ESPs to drive each partition, as mentioned in the @joeflower's post, is the way to go.

But if you want to pursue a single ESP solution, there is a hack that may help. In the WS2812FX.cpp file (line 68) you'll find the service function:
```c++
void WS2812FX::service() {
  if(_running || _triggered) {
    unsigned long now = millis(); // Be aware, millis() rolls over every 49 days
    bool doShow = false;
    for(uint8_t i=0; i < _num_segments; i++) {
      _segment_index = i;
      CLR_FRAME;
      if(now > SEGMENT_RUNTIME.next_time || _triggered) {
        SET_FRAME;
        doShow = true;
        uint16_t delay = (this->*_mode[SEGMENT.mode])();
        SEGMENT_RUNTIME.next_time = now + max(delay, SPEED_MIN);
        SEGMENT_RUNTIME.counter_mode_call++;
      }
    }
    if(doShow) {
      delay(1); // for ESP32 (see https://forums.adafruit.com/viewtopic.php?f=47&t=117327)
      show();
    }
    _triggered = false;
  }
}
```

To run the animations faster, you can run the effect function more then once, like so:
```c++
        doShow = true;
        uint16_t delay = (this->*_mode[SEGMENT.mode])();
        delay = (this->*_mode[SEGMENT.mode])(); // run twice to double animation speed
        SEGMENT_RUNTIME.next_time = now + max(delay, SPEED_MIN);
```

Running the function twice will double the animation speed, running it three times will triple the animation speed, etc. It's crude, but it works.

moose4lord, 2019-02-22:
Oops, I forgot to mention the Teensy and OctoWs2811 solution at [pjrc.com](https://www.pjrc.com/store/octo28_adaptor.html). Although I have never used this particular hardware setup (and I'm not sure if McLighting supports it), I like the design and if I were to pursue a very large installation, I think I would go that route. Just my 2 cents.

DennisH1978, 2019-03-04:
Hi moose4lord...i tried this with double running the effect function. It it a little bit faster, but when double running, the programm just uses every second LED. by running three times the effects function the programm uses every third LED.

moose4lord, 2019-03-04:
That's right. Running the effect function twice simply moves the pixel twice, then displays it (for dual scan effect). So it's effectively skipping animation frames in order to run the animation faster. Usually with so many LEDs, viewed from a distance, it doesn't look too bad.

Another option is to divide you strip into sections and drive all sections in parallel. That would let you fun the animation faster, but of course all sections of the strip would run the same animation.

Other than that, I can't think of anything else to try. You're kinda stuck with the slow serial protocol of the ws28xx LEDs. You can get much higher frame rates with clocked RGB LEDs, like the APA102/DotStar.

DennisH1978, 2019-03-04:
Debashish Sahu wrote me this:

@DennisH1978 <https://github.com/DennisH1978> speed is relative to the
number of pixels. We use a non linear scaling function to map 0-256 to
0-65k
https://github.com/toblum/McLighting/blob/1578cafbf0aee2cbec9e5e1b76be3a6754072837/Arduino/McLighting/request_handlers.h#L96
feel
free to change this line to your liking

—

Any ideas? I changed and changed but there was no difference

debsahu, 2019-03-04:
I thought that the mapping function could be customized for larger number of LEDs. But at the highest value the speed is still slow. So nothing can be done in a easy manner to speed up the animations, except the suggestions by others here.

moose4lord, 2019-03-04:
I agree with @debsahu. There's nothing wrong with the speed mapping calculation (although it doesn't have a lot of granularity on the high end, but when you're mapping 256 values to 65536 values, that can't be helped.) When speed is set to 255, the animation is running as fast as it can.

debsahu, 2019-03-07:
Closing, all possible solution have been proposed here. Thanks everyone!

closed 2019-03-07 by debsahu