toblum/McLighting #330: Accessory HomeKit

arturodiazad1, 2019-02-12:
hello, there is the possibility of creating an accessory for homekit without the need for a homebridge, using the homekit libraries ESP-HomeKit from @MaximKulkin for ESP-OPEN-RTOS. And it uses the OTA update system Life-Cycle-Manager (LCM) from @HomeACessoryKid, just like ravenSystem does https://github.com/RavenSystem/esp-homekit-devices

debsahu, 2019-02-13:
Unfortunately this code is not written in RTOS manner. I dont have the expertise to do this, don't know about @toblum. It looks like a wontfix label. You can use a $5 RPI0 to emulate a Homebridge. You can use something like https://www.npmjs.com/package/homekit2mqtt to emulate a homebridge based on MQTT topics. I used to use it until HomeAssistant got Homekit plugin.

toblum, 2019-02-16:
Hi @arturodiazad1,
this functionality would be great to have, but I'm afraid that I'm also not able to rewrite that code so that it fits into McLighting.
But have a look at homekit2mqtt or homebridge. That works quits good.

Regards
Tobias

closed 2019-02-16 by toblum