toblum/McLighting #336: Visual Studio Code  / PlatformIO  ERROR ✓   I  Like a little help!

lampenschakelen, 2019-02-15:
Hi @toblum Thanks for youtube video: https://youtu.be/7axaXO8cFo8?list=PL3RrveYYJCuRDhrZXcRSQgtsIZ2lH64bv

I am using a Wemos D1 but don't get the WS2812 to work.

https://pastebin.com/nD5184tb 

Here the error: 
[image]
`
I hope somebody can help me out!

Best Regards, 

Klaas

toblum, 2019-02-16:
Hi @lampenschakelen,

this shouldn't happen and when I try it out it works without problems. Did you use the latest version from the master branch? Try to call "clean" (bin icon) and recompile using the hook icon.

Regards
Tobias

debsahu, 2019-02-17:
Looks like you installed VSCode and there is an additional step of installing PlatformIO extension.

lampenschakelen, 2019-02-19:
Today i did fresh installation on my desktop! It's working without any problems now. 
When i have time i will double check it on my laptop. @toblum Thanks a for making the video. 
@debsahu  i did also the installation of PlatformIO extension.

toblum, 2019-02-20:
Great! Good to hear.

closed 2019-02-20 by toblum