toblum/McLighting #337: Wifi not displaying at home

Sreemagic, 2019-02-17:
I did full step and upload the code on nodemcu esp8266 from my office pc and its shows shows the wifi deatails while searching. But the issue is if i connect same device at home no wifi is displaying. Even i re uploaded the script from my home using laptop still not working, its shows a message that uploading done. 
Also issue with multi client one ArduinoJson u installed all the version 6.0  to latest still shows not compatible

debsahu, 2019-02-17:
What is your output of serial console? The code tells you the exact version of ArduinoJSON to install. Remove all instances/folders of ArduinoJSON from your libraries folder of Arduino IDE.

Sreemagic, 2019-02-18:
Thanking you, reinstalled all the software and re uploaded all the code, then its start working

closed 2019-02-18 by Sreemagic