toblum/McLighting #340: AMQTT --> LWT (last will and testament)

SmilyHM, 2019-02-22:
Hi everyone,

I would like to set up a a lwt-topic (with retain) 
/.../.../online
--> connect = true
lwt = false

Is there a way to define it or is it a todo? I could not find it...

Thanks + best regards
Sascha

debsahu, 2019-02-22:
Not in our Todo. I'll have to read up on it's implemention and use. Looks like one sends a message on MQTT connect with retain flag.

debsahu, 2019-02-22:
Async MQTT
https://github.com/marvinroger/async-mqtt-client/blob/master/docs/2.-API-reference.md#asyncmqttclient-setwillconst-char-topic-uint8_t-qos-bool-retain-const-char-payload--nullptr-size_t-length--0

PubSubClient
https://pubsubclient.knolleary.net/api.html#connect5

debsahu, 2019-03-05:
@SmilyHM Could you check the LWT implementation? It is the develop branch, we will be merging with master soon.

SmilyHM, 2019-03-08:
> @SmilyHM Could you check the LWT implementation? It is the develop branch, we will be merging with master soon.

Hi! Of course --> I think I got it done until Saturday!

SmilyHM, 2019-03-08:
@debsahu: lt works :)

Only at "definitions.h" it should be changed:

const char mqtt_will_payload[] = "**false**";      (or "OFFLINE")
instead of const char mqtt_will_payload[] = "**ONLINE**";

The will_payload is the message, the mqtt-boker broadcasts, if the mqtt-client is dead --> so it should be "OFFLINE" or "false".

I am using FHEM: The reading stays "OFFLINE" until a new message "OFFLINE" is send from the broker.

For example "Shelly"-Devices are sending a "true" on first connect to the broker an the lwt is "false".
Or could this be "hardcoded"? --> on 1st connection to the broker a message "true" is send using the "mqtt_will_topic"?

Best regards
Sascha

debsahu, 2019-03-08:
Thanks Sacha, now it all makes sense. 

What do you mean by first time? On boot? Or first configuration ever?

debsahu, 2019-03-08:
Check develop, I only did the on-boot action to send ONLINE to mqtt_will_topic and broker will take care of sending OFFLINE if there are disgraceful MQTT disconnection.

SmilyHM, 2019-03-08:
Hi @debsahu,

I meant an "alive" with each first connection establishment to the broker -> that you already have correctly understood and implemented :-)

It works fine now! Thank you for this upgrade! :-)

Screenshots (FHEM):

"PowerOn / connect to broker (=alive)"
[image]

"PowerOff without sign-off from broker (=dead)"
[image]

Best regards
Sascha

debsahu, 2019-03-16:
@SmilyHM Quick question, why the first MQTT connect message should be "ONLINE"? Shouldn't for subsequent reconnects ESP8266 send "ONLINE" MQTT message?

SmilyHM, 2019-03-22:
@debsahu 

Hi, I'm not exactly sure what you mean. When starting, "ONLINE" is set. After a restart, "ONLINE" is set again. That's ok for me.

Or do you mean, if the MQTT Broker was offline and the ESP connects again? I did not test this. Basically, the MQTT broker should be consistently available - otherwise there is a bigger problem. For me personally, it would not be a big problem to restart the devices once. If it is not a big problem to send the lwt every time you connect to the broker, that would of course be the best solution.

Best regards
Sascha

closed 2019-05-01 by debsahu