toblum/McLighting #341: Do not receive MQTT send

Obicom, 2019-02-23:
I tried today the current version of McLighting and is really a great work.
I would like to implement into Home Assistant and setup the definition.h as descriped.
On the first view all lokks good ... but if I send a mqtt in topic I get no topic back for out.

**Serial monitor shows:**
21:31:44.785 -> MQTT: Message arrived [{"mode":1,"strip_mode":1,"brightness":196,"speed":196,"red":128,"green":128,"blue":128}]
21:31:47.779 -> Saving cfg: {"mode":1,"strip_mode":11,"brightness":196,"speed":196,"red":255,"green":0,"blue":0} Success!
21:31:49.760 -> MQTT: Send [home/McLighting01_ha/state/out]: {"state":"OFF","color":{"r":255,"g":0,"b":0},"brightness":196,"color_temp":327,"speed":196,"effect":"Rainbow"}
21:31:49.795 -> Heap size: 17320

But Mqtt monitor doesn't receive the out topic. Any idea what I could do or check?
[image]

Obicom, 2019-02-23:
Update: I changed now from MQTT to AMQTT  and now it works.

debsahu, 2019-02-23:
Have to change to `#define MQTT_MAX_PACKET_SIZE 512` in PubSubClient.h

closed 2019-02-23 by debsahu

Obicom, 2019-02-23:
Thank you for the quick answer. I will try ASAP. Do you have also a idea why auto discovering not work with  Mosquitto brocker addon in Hassio? Do it work only with the embedded brocker?

`mqtt:
  discovery: true
  discovery_prefix: homeassistant
  broker: 192.168.1.86
  port: 1883
  client_id: home-assistant
  keepalive: 60`

debsahu, 2019-02-24:
There are breaking changes in 0.87 onwards, please try the [develop version](https://github.com/toblum/McLighting/tree/develop). I have to test the develop version this weekend and will push it to master soon.

Obicom, 2019-02-24:
I am still testing the developer version with the same result.