toblum/McLighting #342: [Question] Could you explain how to test new features in developers version?

Obicom, 2019-02-24:
Could you be so kind to explain how I can test the new features in the developer version?

[image]

debsahu, 2019-02-24:
Check version_info.ino for exact info, copying those info here

* new "REST API": /pixels?ct=xxx to change length of LED strip
* new "REST API": /pixels?rgbo=xxx to change RGB order
* new "REST API": /pixels?pin=GPIO_NO to change PIN# (Allowed GPIO values: 16/5/4/0/2/14/12/13/15/3/1)
* In definitions.h, `//#define USE_HTML_MIN_GZ      //uncomment for using index.htm & edit.htm from PROGMEM instead of SPIFFs`
* Use E1.31 mode
* "/esp_status" and "/status" REST API give a lot of information on the build

Obicom, 2019-02-24:
Thank you very much for the parts of information. For music theme I do not understand how E1.31 works but will do some research but would it possible to implement a microphone shield to do in future? I looking forward to see this features on the web GUI ... this would be awesome but step by step ;)

debsahu, 2019-02-24:
Wiki will be updated when we merge to master. Web interface update is far away as of now. You can watch this video I created on how to get started with E1.31: https://youtu.be/lZ09GlO2_8s

Obicom, 2019-02-24:
Thank you very much ... found this video and have watched. Well done. What I am curious of and would like to ask .. why is it so hard to implement the setup for Pixel, RGB Order and Pin setup to the initial WiFi & MQTT setup? I found the neoconfig.json file with the values and as a noob would say will be easy doing to change the values via web interface. But as I mentioned .. I am a noob ;)

Obicom, 2019-02-24:
Do you know Andrey Hübner? He did a comparable project and on the first view (of his videos) I would say he work with a (modified?) mclightning and have the Pin & Co setup plus microphone input in his solution.

https://www.youtube.com/watch?v=gsBx_jLck88

debsahu, 2019-02-25:
It's not hard, but rather not a priority. As a number of people have asked for it, the idea to use pointers was implemented. Now one can change these parameters on the fly. In a few releases, we will take a stab at improving the web interface. The idea to include these parameters in WiFimanager is great, we might do that soon. Adding microphone is not going to happen unless someone is brave enough to tackle it in such a way that does not effect the performance. I have received a MSGEQ7 chip recently, might try playing with sound and light. No promises, and it might happen that it may not get implemented here.

debsahu, 2019-03-07:
Wiki will be updated on how to use these.

closed 2019-03-07 by debsahu