toblum/McLighting #345: [Question] How to use Custom Effects in 2.2.1?

Obicom, 2019-02-28:
I read a lot of topic here in regards for custom effects but to be honest I didn't understand it. Do the new version 2.2.1 use WS2812FX or NeoPixel library? How can I use the custom effects 56-59 with predefined effects from WS2812 custom folder? Would it be possible that someone explain is step by step for a noob?
Thank you very much in advance ....

debsahu, 2019-02-28:
See if this helps? https://github.com/toblum/McLighting/issues/328

debsahu, 2019-02-28:
And replace **strip.** with **strip->**

Obicom, 2019-02-28:
will this work to use all four free custom effect slots or only for one?

debsahu, 2019-02-28:
I haven't tried more than one animation yet, but nothing should be different from WS2812FX examples. 

@moose4lord To add more than one custom animation, one can just call `strip->setCustomMode(myCustomEffect);` multiple times with different animations?

Obicom, 2019-02-28:
I got a lot of error if I did as explained in your first post:

> In file included from D:\Downloads\ESP8266\McLighting-develop\Arduino\McLighting\McLighting.ino:12:0:

D:\OneDrive\Dokumente\Arduino\libraries\FastLED/FastLED.h:14:21: note: #pragma message: FastLED version 3.002.006

 #    pragma message "FastLED version 3.002.006"

                     ^

In file included from D:\Downloads\ESP8266\McLighting-develop\Arduino\McLighting\McLighting.ino:1:0:

definitions.h:6:13: error: expected '>' before numeric constant

 #define PIN 14           // PIN (14 / D5) where neopixel / WS2811 strip is attached

             ^

D:\OneDrive\Dokumente\Arduino\libraries\FastLED/fastpin.h:204:18: note: in expansion of macro 'PIN'

 template<uint8_t PIN> class FastPin {

                  ^

definitions.h:6:13: error: expected '>' before numeric constant

 #define PIN 14           // PIN (14 / D5) where neopixel / WS2811 strip is attached

             ^

D:\OneDrive\Dokumente\Arduino\libraries\FastLED/fastpin.h:239:18: note: in expansion of macro 'PIN'

 template<uint8_t PIN> class FastPinBB : public FastPin<PIN> {};

                  ^

In file included from D:\OneDrive\Dokumente\Arduino\libraries\FastLED/FastLED.h:48:0,

                 from D:\Downloads\ESP8266\McLighting-develop\Arduino\McLighting\McLighting.ino:12:

D:\OneDrive\Dokumente\Arduino\libraries\FastLED/fastpin.h: In instantiation of 'class FastPin<14u>':

D:\OneDrive\Dokumente\Arduino\libraries\FastLED/fastpin.h:239:48:   required from here

D:\OneDrive\Dokumente\Arduino\libraries\FastLED/fastpin.h:207:2: error: static assertion failed: Invalid pin specified

  static_assert(validpin(), "Invalid pin specified");

  ^

In file included from D:\Downloads\ESP8266\McLighting-develop\Arduino\McLighting\McLighting.ino:1:0:

definitions.h:6:13: error: expected '>' before numeric constant

 #define PIN 14           // PIN (14 / D5) where neopixel / WS2811 strip is attached

             ^

D:\OneDrive\Dokumente\Arduino\libraries\FastLED/platforms/esp/8266/fastpin_esp8266.h:14:18: note: in expansion of macro 'PIN'

 template<uint8_t PIN, uint32_t MASK> class _ESPPIN {

                  ^

In file included from D:\OneDrive\Dokumente\Arduino\libraries\FastLED/platforms/esp/8266/fastled_esp8266.h:3:0,

                 from D:\OneDrive\Dokumente\Arduino\libraries\FastLED/platforms.h:29,

                 from D:\OneDrive\Dokumente\Arduino\libraries\FastLED/FastLED.h:52,

[... 628 lines cut ...]
D:\OneDrive\Dokumente\Arduino\libraries\FastLED/platforms/esp/8266/clockless_block_esp8266.h:21:7: error: provided for 'template<unsigned char LANES, int FIRST_PIN, int T1, int T2, int T3, EOrder RGB_ORDER, int XTRA0, bool FLIP, int WAIT_TIME> class InlineBlockClocklessController'

 class InlineBlockClocklessController : public CPixelLEDController<RGB_ORDER, LANES, PORT_MASK> {

       ^

D:\Downloads\ESP8266\McLighting-develop\Arduino\McLighting\McLighting.ino: In function 'uint16_t myCustomEffect()':

McLighting:279:17: error: request for member 'getSpeed' in 'strip', which is of pointer type 'WS2812FX*' (maybe you meant to use '->' ?)

   return (strip.getSpeed() / NUMLEDS);

                 ^

D:\Downloads\ESP8266\McLighting-develop\Arduino\McLighting\McLighting.ino: In function 'void Fire2012()':

McLighting:350:11: error: 'gReverseDirection' was not declared in this scope

       if( gReverseDirection ) {

           ^

McLighting:358:7: error: 'ws2812fx' was not declared in this scope

       ws2812fx.setPixelColor(pixelnumber, color.red, color.green, color.blue);

debsahu, 2019-02-28:
Try this branch out: https://github.com/toblum/McLighting/tree/fastled_custom_effects changes done to 2.2.1 are listed here: https://github.com/toblum/McLighting/commit/e3f86fe80598f3958dc249320063c1d297230442

still only custom animation is done, don't know how to add more than one.

Obicom, 2019-02-28:
Thank you ... I will try it tomorrow ... but must learn is not a "easy thing" to implement additional custom effects.

Obicom, 2019-03-01:
I have tried your sketch but during compilation I got a lot of errors:

> In file included from D:\Downloads\ESP8266\McLighting-fastled_custom_effects\Arduino\McLighting\McLighting.ino:230:0:

request_handlers.h:1457:19: error: 'LED_PIN' was not declared in this scope

     uint8_t pin = LED_PIN;

                   ^

In file included from D:\Downloads\ESP8266\McLighting-fastled_custom_effects\Arduino\McLighting\McLighting.ino:232:0:

D:\OneDrive\Dokumente\Arduino\libraries\FastLED/FastLED.h:14:21: note: #pragma message: FastLED version 3.002.006

 #    pragma message "FastLED version 3.002.006"

                     ^

In file included from D:\Downloads\ESP8266\McLighting-fastled_custom_effects\Arduino\McLighting\McLighting.ino:1:0:

definitions.h:6:13: error: expected '>' before numeric constant

 #define PIN 14           // PIN (14 / D5) where neopixel / WS2811 strip is attached

             ^

D:\OneDrive\Dokumente\Arduino\libraries\FastLED/fastpin.h:204:18: note: in expansion of macro 'PIN'

 template<uint8_t PIN> class FastPin {

                  ^

definitions.h:6:13: error: expected '>' before numeric constant

 #define PIN 14           // PIN (14 / D5) where neopixel / WS2811 strip is attached

             ^

D:\OneDrive\Dokumente\Arduino\libraries\FastLED/fastpin.h:239:18: note: in expansion of macro 'PIN'

 template<uint8_t PIN> class FastPinBB : public FastPin<PIN> {};

                  ^

In file included from D:\OneDrive\Dokumente\Arduino\libraries\FastLED/FastLED.h:48:0,

                 from D:\Downloads\ESP8266\McLighting-fastled_custom_effects\Arduino\McLighting\McLighting.ino:232:

D:\OneDrive\Dokumente\Arduino\libraries\FastLED/fastpin.h: In instantiation of 'class FastPin<14u>':

D:\OneDrive\Dokumente\Arduino\libraries\FastLED/fastpin.h:239:48:   required from here

D:\OneDrive\Dokumente\Arduino\libraries\FastLED/fastpin.h:207:2: error: static assertion failed: Invalid pin specified

  static_assert(validpin(), "Invalid pin specified");

  ^

In file included from D:\Downloads\ESP8266\McLighting-fastled_custom_effects\Arduino\McLighting\McLighting.ino:1:0:

definitions.h:6:13: error: expected '>' before numeric constant

 #define PIN 14           // PIN (14 / D5) where neopixel / WS2811 strip is attached

             ^

D:\OneDrive\Dokumente\Arduino\libraries\FastLED/platforms/esp/8266/fastpin_esp8266.h:14:18: note: in expansion of macro 'PIN'

 template<uint8_t PIN, uint32_t MASK> class _ESPPIN {

[... 624 lines cut ...]
     case UCS1903_PORTA: return addLeds(new InlineBlockClocklessController<NUM_LANES, PORTA_FIRST_PIN, NS(500), NS(1500), NS(500), RGB_ORDER>(), data, nLedsOrOffset, nLedsIfOffset);

                                                                                                                                            ^

In file included from D:\OneDrive\Dokumente\Arduino\libraries\FastLED/platforms/esp/8266/fastled_esp8266.h:5:0,

                 from D:\OneDrive\Dokumente\Arduino\libraries\FastLED/platforms.h:29,

                 from D:\OneDrive\Dokumente\Arduino\libraries\FastLED/FastLED.h:52,

                 from D:\Downloads\ESP8266\McLighting-fastled_custom_effects\Arduino\McLighting\McLighting.ino:232:

D:\OneDrive\Dokumente\Arduino\libraries\FastLED/platforms/esp/8266/clockless_block_esp8266.h:21:7: error: provided for 'template<unsigned char LANES, int FIRST_PIN, int T1, int T2, int T3, EOrder RGB_ORDER, int XTRA0, bool FLIP, int WAIT_TIME> class InlineBlockClocklessController'

 class InlineBlockClocklessController : public CPixelLEDController<RGB_ORDER, LANES, PORT_MASK> {

       ^

D:\Downloads\ESP8266\McLighting-fastled_custom_effects\Arduino\McLighting\McLighting.ino: In lambda function:

McLighting:671:21: error: 'LED_PIN' was not declared in this scope

       json["pin"] = LED_PIN;

                     ^

debsahu, 2019-03-01:
Please download ALL files in that branch. The definition of LED_PIN is in **definitions.h** And everything should compile see logs https://travis-ci.org/toblum/McLighting/builds/500076055

Obicom, 2019-03-01:
Ok, thank you for the advice. I was not aware that you changed something in definitions.h and used my own preconfigured one. Now I get only one error .. but I guess is more a warning ...

`In file included from D:\Downloads\ESP8266\McLighting-fastled_custom_effects\Arduino\McLighting\McLighting.ino:232:0:

D:\OneDrive\Dokumente\Arduino\libraries\FastLED/FastLED.h:14:21: note: #pragma message: FastLED version 3.002.006

 #    pragma message "FastLED version 3.002.006"

                     ^

In file included from D:\OneDrive\Dokumente\Arduino\libraries\FastLED/FastLED.h:65:0,

                 from D:\Downloads\ESP8266\McLighting-fastled_custom_effects\Arduino\McLighting\McLighting.ino:232:

D:\OneDrive\Dokumente\Arduino\libraries\FastLED/fastspi.h:110:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output

 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"`

debsahu, 2019-03-01:
That's not an error, but a warning from FastLED library for esp8266. Shouldn't effect compiling your binary, we use FastLED to do the color calculations and not send any data. WS2812FX sends out the data, so ignore that warning!

Obicom, 2019-03-01:
Thank you very much again ....

debsahu, 2019-03-07:
see latest version to use custom effects.

closed 2019-03-07 by debsahu