toblum/McLighting #349: Homeassistant start entity ever in state static

Teckware, 2019-03-04:
Hello Tobias and all involved, First of all, a big compliment for your programming. But I have a little problem. I have managed the implementation in my home assistant successfully with them, but when I start the individual McLigting's, I have now built 16 pieces in the house, they always begin with effect = static. If I start you single, then you always take the last state and save the previous running "effect-state" not so when starting on the home assistant there always start in the "effect = static" So here's my question: where can I give them one give specific default for a start under Home assistant. And also the speed I would like to be able to adjust to the Home assistant.

debsahu, 2019-03-04:
I see what you are alluding to, all we have to do is send the state MQTT message on MQTT connect. Easy to-do.

Teckware, 2019-03-04:
Hi debsahu,
In the meantime, I have found a solution ... Nearly.
I start with node red (pic-2) since I can start the individual flow it as I want. And if I want to start individual lightings I do that with the HA (pic-1), and if I'm quite lazy I start everything by google home assistant. But would still be nice if there could be a speed slider in pic-3.

And please carry on like this!!

Teckware
[image]
[image]
[image]

debsahu, 2019-03-04:
Update is pushed with this on develop branch. This issue will close as soon as we merge the develop to master branch.

Teckware, 2019-03-04:
oki

debsahu, 2019-03-05:
Tested this and it seems to be working. The stateOn was not being updated upon reading animation settings from SPIFFs, should be fixed in the latest version.

closed 2019-03-07 by debsahu