toblum/McLighting #351: Using 12v ws2811, only first IC lights up.

freudianSLAP, 2019-03-07:
Hello,

I'm stuck. I managed to get everything installed and got the esp8266 flashed and connected to my wifi. I can see commands coming through the serial monitor when I select different modes or colors in the web GUI. However those commands dont seem to affect the 12v ws2811 strip at all. Only the first three LEDs (1 IC) are permanently on blue. They select that at startup and then nothing else happens. Any help would be greatly appreciated, Ive been trying to figure this out nonstop for the last two days and I'm at a loss.

Here is the output from ip-address/esp_status:

"HOSTNAME" "McLighting01"
"version""2.2.0"
"heap":24376,
"sketch_size":414608,
"free_sketch_space":2727936,
"flash_chip_size":4194304,
"flash_chip_real_size":4194304,
"flash_chip_speed":40000000,
"sdk_version":"2.2.1(cfd48f3)",
"core_version":"2_4_2",
"cpu_freq":80,
"chip_id":1458376,
"animation_lib":"WS2812FX_DMA",
"pin":3,
"number_leds":100,
"button_mode":"ON",
"home_assistant":"OFF",
"legacy_animations":"ON"
,"esp8266_http_updateserver":"ON",
"arduino_ota":"ON",
"state_save":"SPIFFS"}

Since I'm using the 12v strip I changed the wiring a little bit, I don't know if that's whats causing the problem. Here's a diagram of how I wired everything up. It starts with a 12v power supply that powers the strip directly. There's a seperate line that gets stepped down to 5v for the esp8266. After that wiring is basically identical to what is in the McLighting hardware tutorial.

[image]

And here is my definitions.h file as I'm currently uploading it to the esp8266

`#define USE_WS2812FX_DMA      // Uses PIN is ignored & set to RX/GPIO3  Uses WS2812FX, see: https://github.com/kitesurfer1404/WS2812FX
//#define USE_WS2812FX_UART1     // Uses PIN is ignored & set to D4/GPIO2  Uses WS2812FX, see: https://github.com/kitesurfer1404/WS2812FX
//#define USE_WS2812FX_UART2     // Uses PIN is ignored & set to TX/GPIO1  Uses WS2812FX, see: https://github.com/kitesurfer1404/WS2812FX

// Neopixel
#define PIN D2           // PIN (14 / D5) where neopixel / WS2811 strip is attached
#define NUMLEDS 100       // Number of leds in the strip
#define BUILTIN_LED 2    // ESP-12F has the built in LED on GPIO2, see https://github.com/esp8266/Arduino/issues/2192
#define BUTTON 4         // Input pin (4 / D2) for switching the LED strip on / off, connect this PIN to ground to trigger button.

const char HOSTNAME[] = "McLighting01";   // Friedly hostname

#define HTTP_OTA             // If defined, enable ESP8266HTTPUpdateServer OTA code.
#define ENABLE_OTA         // If defined, enable Arduino OTA code.
//#define ENABLE_AMQTT         // If defined, enable Async MQTT code, see: https://github.com/marvinroger/async-mqtt-client
//#define ENABLE_MQTT        // If defined, enable MQTT client code, see: https://github.com/toblum/McLighting/wiki/MQTT-API
//#define ENABLE_HOMEASSISTANT // If defined, enable Homeassistant integration, ENABLE_MQTT or ENABLE_AMQTT must be active
#define ENABLE_BUTTON        // If defined, enable button handling code, see: https://github.com/toblum/McLighting/wiki/Button-control
//#define MQTT_HOME_ASSISTANT_SUPPORT // If defined, use AMQTT and select Tools -> IwIP Variant -> Higher Bandwidth
#define ENABLE_LEGACY_ANIMATIONS // Dont disbale this for now
#define ENABLE_E131              // E1.31 implementation

#ifdef ENABLE_E131
  #define START_UNIVERSE 1                    // First DMX Universe to listen for
  #define END_UNIVERSE 2              // Total number of Universes to listen for, starting at UNIVERSE
#endif

//#define WIFIMGR_PORTAL_TIMEOUT 180
//#define WIFIMGR_SET_MANUAL_IP

#ifdef WIFIMGR_SET_MANUAL_IP
  uint8_t _ip[4] = {192,168,0,128};
  uint8_t _gw[4] = {192,168,0,1};
  uint8_t _sn[4] = {255,255,255,0};
#endif

[... 111 lines cut ...]
  char current_state[32];               // Keeps the current state representation
  char last_state[32];                  // Save the last state as string representation
  unsigned long time_statechange = 0;   // Time when the state last changed
  int timeout_statechange_save = 5000;  // Timeout in ms to wait before state is saved
  bool state_save_requested = false;    // State has to be saved after timeout
#endif
#ifdef ENABLE_STATE_SAVE_SPIFFS
  bool updateStateFS = false;
#endif

// Button handling
#ifdef ENABLE_BUTTON
  #define BTN_MODE_SHORT "STA| 1|  0|245|196|255|255|255"   // Static white
  #define BTN_MODE_MEDIUM "STA| 1| 48|245|196|255|102|  0"    // Fire flicker
  #define BTN_MODE_LONG "STA| 1| 46|253|196|255|102|  0"  // Fireworks random

  unsigned long keyPrevMillis = 0;
  const unsigned long keySampleIntervalMs = 25;
  byte longKeyPressCountMax = 80;       // 80 * 25 = 2000 ms
  byte mediumKeyPressCountMin = 20;     // 20 * 25 = 500 ms
  byte KeyPressCount = 0;
  byte prevKeyState = HIGH;             // button is active low
  boolean buttonState = false;
#endif
`

debsahu, 2019-03-07:
Since you have enabled `#define USE_WS2812FX_DMA`, PIN is ignored and you will have to use PIN **RX/GPIO3** and not D2.

Since you are using 2811 strip, you will have to use **NeoEsp8266Dma400KbpsMethod**,

Replace https://github.com/toblum/McLighting/blob/c0329ef01e27cfb267fa41411868e4c441a22a02/Arduino/McLighting/McLighting.ino#L94 with 
```
NeoEsp8266Dma400KbpsMethod* dma;
```

and replace https://github.com/toblum/McLighting/blob/c0329ef01e27cfb267fa41411868e4c441a22a02/Arduino/McLighting/McLighting.ino#L106
```
dma = new NeoEsp8266Dma400KbpsMethod(stripSize, 3);
```

debsahu, 2019-03-07:
I made it easier in [Develop Branch](https://github.com/toblum/McLighting/tree/develop) by just uncommenting following line https://github.com/toblum/McLighting/blob/fa918443c4063d418e25698b695b18549fd4beb0/Arduino/McLighting/definitions.h#L8

freudianSLAP, 2019-03-07:
Thank you so much for getting back to me so quickly, I really appreciate it! So I deleted my McLighting folder and did a new clone of the McLighting repository. Though it seems like my newly cloned definitions.h file didn't have the updates you made, so I manually copy-pasted the raw definitions.h from the git website.

Here is how I have the beginning of your new definitions.h set up (I moved my data pin to d5 and left ws2812fx commented):

```

//#define USE_WS2812FX_DMA       // LED_PIN is ignored & set to RX/GPIO3  Uses WS2812FX, see: https://github.com/kitesurfer1404/WS2812FX
//#define USE_WS2812FX_UART1     // LED_PIN is ignored & set to D4/GPIO2  Uses WS2812FX, see: https://github.com/kitesurfer1404/WS2812FX
//#define USE_WS2812FX_UART2     // LED_PIN is ignored & set to TX/GPIO1  Uses WS2812FX, see: https://github.com/kitesurfer1404/WS2812FX

// Neopixel
#define LED_PIN D5       // LED_PIN (14 / D5) where neopixel / WS2811 strip is attached
#define NUMLEDS 100       // Number of leds in the strip
#define LED_TYPE_WS2811  // Uncomment if LED type uses 400 KHz (classic 'v1' (not v2) FLORA pixels, WS2811 drivers)
#define BUILTIN_LED 2    // ESP-12F has the built in LED on GPIO2, see https://github.com/esp8266/Arduino/issues/2192
#define BUTTON 4         // Input pin (4 / D2) for switching the LED strip on / off, connect this PIN to ground to trigger button.

#define HOSTNAME "McLighting01"   // Friedly hostname

#define HTTP_OTA             // If defined, enable ESP8266HTTPUpdateServer OTA code.
#define ENABLE_OTA         // If defined, enable Arduino OTA code.
//#define ENABLE_AMQTT         // If defined, enable Async MQTT code, see: https://github.com/marvinroger/async-mqtt-client
//#define ENABLE_MQTT        // If defined, enable MQTT client code, see: https://github.com/toblum/McLighting/wiki/MQTT-API
//#define ENABLE_HOMEASSISTANT // If defined, enable Homeassistant integration, ENABLE_MQTT or ENABLE_AMQTT must be active
#define ENABLE_BUTTON        // If defined, enable button handling code, see: https://github.com/toblum/McLighting/wiki/Button-control
//#define MQTT_HOME_ASSISTANT_SUPPORT // If defined, use AMQTT and select Tools -> IwIP Variant -> Higher Bandwidth
#define ENABLE_LEGACY_ANIMATIONS // Dont disbale this for now
#define ENABLE_E131              // E1.31 implementation
#define USE_HTML_MIN_GZ             //comment for using index.htm & edit.htm from SPIFFs instead of PROGMEM
//#define CUSTOM_WS2812FX_ANIMATIONS  //uncomment and put animations in "custom_ws2812fx_animations.h" 
```
WHen I try to build I get this error:

```
Compiling .pioenvs\nodemcuv2\src\McLighting.ino.cpp.o
In file included from C:/Users/Souren/Desktop/McLighting/Arduino/McLighting/McLighting.ino:459:0:
Arduino\McLighting\request_handlers.h: In function 'bool readStateFS()':
Arduino\McLighting\request_handlers.h:1415:25: error: 'stateOn' was not declared in this scope
if(mode != OFF) stateOn = true;
^
*** [.pioenvs\nodemcuv2\src\McLighting.ino.cpp.o] Error 1
```

debsahu, 2019-03-07:
Please use this [ZIP File](https://github.com/toblum/McLighting/archive/develop.zip)

freudianSLAP, 2019-03-08:
Ok I delted everything and loaded the zip you provided. Building it returns this error:

```
hting\request_handlers.h: In function 'bool readStateFS()':
Arduino\McLighting\request_handlers.h:1415:25: error: 'stateOn' was not declared in this scope
if(mode != OFF) stateOn = true;
^
Compiling .pioenvs\nodemcuv2\libe96\AsyncMqttClient_ID346\AsyncMqttClient\Packets\UnsubAckPacket.cpp.o
Archiving .pioenvs\nodemcuv2\lib833\libArduinoJson.a
Compiling .pioenvs\nodemcuv2\lib88f\Adafruit NeoPixel_ID28\Adafruit_NeoPixel.cpp.o
Compiling .pioenvs\nodemcuv2\lib88f\Adafruit NeoPixel_ID28\esp8266.c.o
*** [.pioenvs\nodemcuv2\src\McLighting.ino.cpp.o] Error 1
Compiling .pioenvs\nodemcuv2\lib919\WS2812FX_ID1647\WS2812FX.cpp.o
```

debsahu, 2019-03-08:
Thank you, it looks like I overlooked the fact that stateOn is defined for Home Assistant. For now, could you uncomment `//#define ENABLE_HOMEASSISTANT`. Even if you don't use HA, McLighting will function.

debsahu, 2019-03-08:
Please update to master version v2.2.2, should be fixed. Please feel free to reopen if you still facing issues.

closed 2019-03-08 by debsahu