toblum/McLighting #352: More bugs 2.2.1

iSeven4pda, 2019-03-08:
2.2.1 This is a failure. The size is growing, but nothing new - just the json curve that has nowhere to apply, except how to use what is given. The web interface is also far from ideal and slows down - even the old "Responsive LED Control" had excellent functions. I’m not talking about integration with Blynk (which I also have offline server) instead of many options for mqtt - which would give many more nice features... Now Mc Lighting still hangs - buttons do not work, then they stick at all. I will continue to bluff blynk with my poor knowledge - it's still better than reinventing the wheel in an attempt to disassemble tons of useless code. Sad...

M4lik, 2019-03-08:
I'm sorry, but I don't really understand you problem. You are bothered by the size of the ... of what? The whole project? Just the web interface?

You complain about a lot, without being specific what exactly is wrong.
I know that things are not ideal, but is this a reason to degrade the whole project?

If you have ideas, comments, whatsoever then feel free to open up an issue on a specific topic, or even make a pull request if you have code to contribute :)

Don't get me wrong. I don't try to be mean or want to wave your comment off. It's just that THIS is not helpful as it is not constructive.

closed 2019-03-08 by M4lik

debsahu, 2019-03-08:
@M4lik Maybe we should create an issue template, that will help with assimilation of issues users are facing. Do you have experience in creating an issue template (some examples can be found for NeoPixelBus library or ESP8266-Arduino Core on GitHub)?

M4lik, 2019-03-08:
@debsahu Not that bad of an idea. I will try to come up with one tomorrow. Maybe even a PR template? 🤔