toblum/McLighting #355: More than 3 custom effects slots?

dermodmaster, 2019-03-09:
Hello, how can I add more buttons for custom effects?

debsahu, 2019-03-09:
There are only 4 slots for WS2812FX.

https://github.com/toblum/McLighting/blob/93cf8a919fccc262ac3effe59e83f229989d4640/Arduino/McLighting/McLighting.ino#L296

Call strip->setCustomMode(XXX, F("Custom1"), Function) 4 times. Where XXX is 0 or 1 or 2 or 3.

dermodmaster, 2019-03-09:
Oh well, thank you for quick response.. I asked because I reached the limit ^^ Thought, it was a limit in your repo.
Found https://github.com/kitesurfer1404/WS2812FX/issues/132 where you can modify the library so you can add more slots 👍 
THX for quick response!!!

closed 2019-03-09 by dermodmaster

debsahu, 2019-03-09:
Yes 4 is a good number and be careful in managing memory with many animations.

dermodmaster, 2019-03-09:
I'll thx m8 x)