toblum/McLighting #357: I cannot make it work

scordal, 2019-03-10:
I am a novice in NodeMcu and Arduino things, but I am learning littlñe by little.
I am an electronic freak and I am trying to make this work. Yesterday was omehow possible but since yesterday I cannot finish the installation part.
I have been struggling with this since two days and seems that today there was a code change that makes impossible to install the index.html
The feedback from Serail COM3 is following:

FS Usage: 0/957314 bytes

Coudnt find config.json
Coudnt find config.json
WiFiManager config FS Read failure!
*WM: Adding parameter
*WM: host
*WM: Adding parameter
*WM: port
*WM: Adding parameter
*WM: user
*WM: Adding parameter
*WM: pass
*WM: Adding parameter
*WM: strip_size
*WM: Adding parameter
*WM: led_pin
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 0
Entered config mode
192.168.4.1
McLighting01
*WM: 
*WM: Configuring access point... 
*WM: McLighting01
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
Saving Strip cfg: {"pixel_pount":300,"rgb_order":82,"pin":14}
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Handle root
*WM: Handle root
*WM: Handle root
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Sent config page
*WM: Request redirected to captive portal
[... 227 lines cut ...]
#ifdef ENABLE_STATE_SAVE_EEPROM
  char current_state[32];               // Keeps the current state representation
  char last_state[32];                  // Save the last state as string representation
  unsigned long time_statechange = 0;   // Time when the state last changed
  int timeout_statechange_save = 5000;  // Timeout in ms to wait before state is saved
  bool state_save_requested = false;    // State has to be saved after timeout
#endif
#ifdef ENABLE_STATE_SAVE_SPIFFS
  bool updateStateFS = false;
#endif

// Button handling
#ifdef ENABLE_BUTTON
  #define BTN_MODE_SHORT "STA| 1|  0|245|196|255|255|255"   // Static white
  #define BTN_MODE_MEDIUM "STA| 1| 48|245|196|255|102|  0"    // Fire flicker
  #define BTN_MODE_LONG "STA| 1| 46|253|196|255|102|  0"  // Fireworks random

  unsigned long keyPrevMillis = 0;
  const unsigned long keySampleIntervalMs = 25;
  byte longKeyPressCountMax = 80;       // 80 * 25 = 2000 ms
  byte mediumKeyPressCountMin = 20;     // 20 * 25 = 500 ms
  byte KeyPressCount = 0;
  byte prevKeyState = HIGH;             // button is active low
  boolean buttonState = false;
#endif

debsahu, 2019-03-10:
Could you decrypt the stack? https://github.com/me-no-dev/EspExceptionDecoder

debsahu, 2019-03-10:
Index.html it's already packed in gzip form, so no need to upload anymore you can comment `//#define USE_HTML_MIN_GZ` to use uploaded index.html file in SPIFFs.

scordal, 2019-03-10:
Hi Debsahu:

I did what you ask to comment the line and the stack results are
Decoding stack results
0x40100220: millis at D:\Users\PC\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\core_esp8266_wiring.c line 183
0x40223a04: WS2812FX::mode_static() at D:\Users\PC\Documents\Arduino\libraries\WS2812FX-master\src\WS2812FX.cpp line 527
0x402236b4: WS2812FX::service() at d:\users\pc\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2\bits/stl_algobase.h line 222
0x40223629: WS2812FX::show() at D:\Users\PC\Documents\Arduino\libraries\WS2812FX-master\src\WS2812FX.cpp line 131
0x402104ed: loop() at D:\Users\PC\Downloads\McLighting-master\Arduino\McLighting/McLighting.ino line 1293
0x40228ac4: loop_wrapper() at D:\Users\PC\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\core_esp8266_main.cpp line 125

debsahu, 2019-03-10:
That looks like https://github.com/toblum/McLighting/issues/309 Let me test this. If you have felixibility in wiring, could you use **RX/GPIO3** pin and uncomment `#define USE_WS2812FX_DMA` OR use **D4/GPIO2** and uncomment `#define USE_WS2812FX_UART1`. These are DMA or UART hardware based method (exploits) that do a better job are shifting large number of RGB data.

scordal, 2019-03-10:
Thanks. I could go thru all the steps now, I used the configuration with  D4/GPIO2, I rewired but when I connect the light strip I got only a few led lit in white and 4 in blue. 
I wonder what I did wrong

scordal, 2019-03-10:
I tested al scenarios you proposed with no luck:
- use RX/GPIO3 pin and uncomment #define USE_WS2812FX_DMA: did not work, the led strip lits a few leds in White and 6 in blue.
- use D4/GPIO2 and uncomment #define USE_WS2812FX_UART1: lits all leds in White and nothing else.
- using D5/gpio14 the unit restarts, and it disconnects from internet.

I give up for today, I will continue some other time. Thanks for your help.

scordal, 2019-03-10:
Just for your reference I use a ws2811 led strip and a level shifter from the output to the DIN of the level strip.

debsahu, 2019-03-10:
You have common ground? Could you share your wiring?

Deinjo, 2019-03-15:
Hello,

i have setup the project today, too.
Now I have completely the same issue with the same stack result (same module, same order, same rows)

debsahu, 2019-03-16:
@Deinjo Please share wiring, I'm not able to produce the error on empty esp8266 (no LEDs). Erase flash and upload sketch at (slow) 115200 baud speed.

scordal, 2019-03-23:
debsahU.I made itr work, though I changed the wiring toi a different pin. I realized I was having problema with my Wifi that is saturated with many Wifi Connections and therefore did not start.

HiranChaudhuri, 2019-04-15:
My Wemos D1 mini would also continually reset itself after I entered WIFI configuration.
Then I installed an MQTT broker (Apache ActiveMQ-Artemis), and entered this information together with the WIFI configuration.

Now the device no longer resets. I can see in ActiveMQ that a queue has been created.
But I can neither access the device's homepage, nor the upload page.

debsahu, 2019-04-15:
Are you using the develop version? Of not please do as most of the reset issue is resolved there.

HiranChaudhuri, 2019-04-15:
The constant restarts and the missing stripstate.json file seemed not connected.
I finally got rid of the restarts by turning off a lot of definitions in definitions.h (and thus reducing functionality). Once the controller was stable, I managed to upload index.htm which now gives me the control as demonstrated in the video.

As soon as I enable MQTT the system again gets unstable. AMQTT does not compile for me at all, I may still need to find some of the dependencies.

HiranChaudhuri, 2019-04-15:
> Are you using the develop version? Of not please do as most of the reset issue is resolved there.

This is what I will have to track better now. Right now I just wanted to get it going. Now that I realize McLightning combines a wealth of libraries I will look into what I installed, and how. Also I want to check how hard it is to enhance the color wheel. My Hue lights not only pick color but also saturation.

HiranChaudhuri, 2019-05-04:
So I made a lot of functions work now. Compiled the cloned master branch with all the latest libraries that were available. It seems McLighting works like a charm, and I found only one feature (home assistant autodiscovery) to not work. Documentation of what I did is available on request.

I will open a separate issue for the autodiscovery part.

closed 2019-07-05 by toblum