toblum/McLighting #358: no access point

richyxi, 2019-03-11:
hi, i was trying to run the software, everything cool but the thing is that there no access point for me, im not sure if i missing something, somebody can help? 
awsome proyect btw!

debsahu, 2019-03-11:
I guess you didn't erase flash completely. Your WiFi credentials are stored in a separate place on Flash and can be used to connect to WiFi for following reboots.
[image]

richyxi, 2019-03-11:
Okay, let me try that out men

richyxi, 2019-03-12:
It worked men, thank u so much.
One little thing, it can work with multiple esp's?

debsahu, 2019-03-12:
Yes, use different **HOSTNAME** for each esp8266. You can use https://www.youtube.com/watch?v=T1Ul7iN72lI as a reference.

closed 2019-03-12 by debsahu