toblum/McLighting #359: System reset itself at random times and clears some of the settings.

Springs87, 2019-03-13:
hi all,

I'm running Mclightning on a D1 Mini with approx 100 leds. LED's are 5v with their own PSU

2 issues i've got are as follows.

1. After some time of the lights running they will reset themselves. there is no set time but it does seem that it's the D1 mini and not the leds

2. when the lights reset themselves the settings i've set don't stick. It will keep the current animation but  the Speed will reset back to standard.

Any advice would be great. I do like the LEDS not their setup as i want them but the issues make it annoying at times.

debsahu, 2019-03-13:
What is your output of http://[IP_ADDRESS]/esp_status

The following line should read the stored speed settings
https://github.com/toblum/McLighting/blob/93cf8a919fccc262ac3effe59e83f229989d4640/Arduino/McLighting/request_handlers.h#L1419
Do you happen to have the output from Serial?

Springs87, 2019-03-15:
Hi,

Finally was able to get the info.

From esp_status: (This was taken after it reset itself)

{"HOSTNAME":"McLighting01","version":"2.1.11","heap":25960,"sketch_size":443072,"free_sketch_space":2699264,"flash_chip_size":4194304,"flash_chip_real_size":4194304,"flash_chip_speed":40000000,"sdk_version":"2.2.1(cfd48f3)","core_version":"2_4_2","cpu_freq":80,"chip_id":1458208,"animation_lib":"WS2812FX","pin":2,"number_leds":123,"button_mode":"ON","amqtt":"ON","home_assistant":"ON","legacy_animations":"ON","esp8266_http_updateserver":"ON","state_save":"SPIFFS"}

debsahu, 2019-03-16:
Could you upgrade to latest version?

donotoverdriveit, 2019-07-11:
Hello, I have the same problem. 

After a while, the system restarts. The stripes then glow static red. Sometimes it turns on, even though it was off before. I use the version 2.2.4

toblum, 2019-07-13:
@donotoverdriveit Do you see anything in the serial output when the restart occurs?