toblum/McLighting #360: WLAN Connection lost

DennisH1978, 2019-03-13:
Hello there,
i got now 4 NodeMCU's (every Node has 120 LED's and a power supply 10A) with McLighting on it. Very happy in fact, but............
2 of them lost their WLAN Connection up to ten times a day.
All 4 Nodes are made with the same sketch, only other IP's and friendly names. So there is no prob, too.
Resistance and capacitor are build in. Every time the connection is lost, the stripe blinks at the first 10 or 15 LED's. What can i do?

debsahu, 2019-03-13:
So weird, I guess it is not software then. If two work flawlessly and two are acting up, I would suggest that it might be an issue with WiFi router (some routers drop clients is strength is low) or the ESP themselves or the power supply to those strips. Try interchanging them and eliminate each one of cases.

DennisH1978, 2019-03-18:
Hello...i just tried different things...i put the sketch on a wemos d1 mini. 1,5 days without any faults. since the third day 20 losts of connection a day. i tried antother supply, no changes. i tried just on an usb port, no changes. since yesterday the wemos is in the same room with the fritzbox, connections fails often a day. and not every 15 minutes or 30 or always in the same time, it is very different. so once 4 fails per hour, once hours nothing. so at least it is not the voltage, it it not the wlan, it is not the  sketch. is there maybe a sleep mode in the sketch or something?

DennisH1978, 2019-03-26:
Update....i got a wemod d1, since the first power on there is no connection lost. and i got 2 other wemos d1, same sketch, just other IP or DHCP, and there is every 15 minutes connection lost. Even if i connect them directly per usb in the router theconnection gets lost. i tried a wemos d1 pro with extra antenna, connection lost. i don't know what to do else to find the issue.

debsahu, 2019-03-26:
Can you use esp8266 Arduino core 2.4.2? There seems to be stability issues with 2.5.0 as per the pinned issue on arduino-esp8266 repo. It effects certain mac/esp8266

DennisH1978, 2019-03-26:
Hello and thank you... but i use core 2.4.2. :-) any other idea? ;-)

HiranChaudhuri, 2019-06-01:
I also perceive McLighting reboots (and non-reachability from the rest of the network). This started after one week that the strip was installed. Right now I suspect two items that are unrelated to McLighting:

- my Wifi router has shown some small glitches before - it could be related. Not sure if the ESP8266 supports 5 GHz, but the 2.4 GHz coverage is not good
- When I touched the Wemos D1 mini board recently I noticed the ESP is quite warm. I wrapped it with tape to lower the builtin blue LED's light emission (after all I wanted to see my neopixels rather than the standard light) but maybe the chip is covered in a cooling case that now no longer works.

I'll followup on those two items. Maybe others have similar symptoms?