toblum/McLighting #361: Question: [RGB-W] Can you the SK6812 fork integrate in Main Projekt?

gitpower2017, 2019-03-14:
Hello,

The main project is much better maintained and debugged.
Meanwhile I can well imagine that the RGBW LED strips are just as popular as the pure RGB LED strips.

ChristophA, 2019-03-14:
I‘d like that, too!

TheAndi, 2019-03-15:
+1
As McLighting is intended as a room/mood lighting solution, I think it should definitely support the SK6812 strips because they can produce much nicer light.

Unfortunately I can't get the FabLab-Version to work properly (https://github.com/FabLab-Luenen/McLighting/issues/4). Therefore, I would really appreciate an official integration.

debsahu, 2019-03-15:
@BPoH_Voodoo is a great collaborator and a programmer. McLighting's porting to RGBW is done single handedly and issues will arise as more things gets added. Everyone here is doing this on a voluntary basis and when free is available. Please give all developers more time to sort through issues, unit test and if possible contribute by coding or help solve an issue marked "help wanted". As such there is no plan to port RGBW strip to McLighting, we could just take this request and mark it as Todo.

vemeT5ak, 2019-03-15:
I'm not sure if this will help anyone. I have added a fork for McL, as of
two weeks ago, that works with the SK2818 RGBW LEDs (
https://github.com/vemeT5ak/McLighting). I have also noted at the top of
the definitions.h and mcL_vemet5ak_edits.ino which lines you will have to
edit to get them to work for your application, at least the stuff I had to
change to get it to work for my application. I may be missing a few files.
If you get a file missing error, take it from the original McLightning
repo.

hope this helps

vemeT5ak, 2019-03-15:
Apologies, I didn't hit "reply all" a good habit I do believe.

I'm not sure if this will help anyone. I have added a fork for McL, as of
two weeks ago, that works with the SK6812 RGBW LEDs (
https://github.com/vemeT5ak/McLighting). I have also noted at the top of
the definitions.h and mcL_vemet5ak_edits.ino which lines you will have to
edit to get them to work for your application, at least the stuff I had to
change to get it to work for my application. I may be missing a few files.
If you get a file missing error, take it from the original McLightning
repo.

hope this helps

debsahu, 2019-03-15:
Thanks @vemeT5ak, will try to add your branch to README

toblum, 2019-03-15:
Sure, having RGBW mode would be great and this is definitely something we could have a look at, but it has many implications as the API has to change and we have to take care that we don't break existing projects and still support RGB. Apart from the API the UI has to adapt to both RGB and RGBW. 
This is all possible but it takes time (which I personally have far too little of at the moment unfortunately) as @debsahu pointed out.
I'll keep all that in mind for the next version of the UI. In the meantime give @bpohvoodoo version a try.
BTW: I ordered a SK6812 RGBW strip some time ago but I fear that I received a RGB version. Maybe some kind of fake product or wrong listing. Is this possible? Are there RGB version of the SK6812 strips out there?

gitpower2017, 2019-03-15:
Hi toblum,

on this picture you can see the different models
http://www.witop-tech.com/wp-content/uploads/2017/06/sk6812-programmable-led-chip-1.jpg

I have the model on the lower right

gitpower2017, 2019-03-15:
@vemeT5ak 

I'm afraid I can't do anything with your explanation.
Can you provide a version that I can compile directly?

TheAndi, 2019-03-15:
Maybe a simple RGB -> RGBW conversion before the data is sent to the strip would do the trick for now. 

There is already a lot of existing code for that like:
http://codewelt.com/rgbw
or 
https://stackoverflow.com/questions/40312216/converting-rgb-to-rgbw

**This would avoid any changes of the API or the UI**. Also, the effects should work seamlessly with that.
It would "only" require a change at the pixel output part and an option in the definition.h for enabling it.

@toblum could be the case that there are just RGB versions concerning that datasheet: http://www.szledcolor.com/download/SK6812%20LED.pdf

Keep up the great work and no pressure ;)

bpohvoodoo, 2019-03-17:
All changes implemented from 2.2.2 and many more!
Html UI is intergrated now, as in original project
Edit.html.gz is integrated now, too.
Homeassistant is now using the white channel, too.
All settings are configurable via rest api, websockets and html ui.
all RGB(W) Orders are supported, can be changed during runtime.
Ui regognizes if W channel is RGB Order, no switch needed anymore.
Many more changes. Read version_info
if @toblum and debsahu want to merge, we can do so.

debsahu, 2019-03-17:
@bpohvoodoo There are a few minor bug fixes I caught yesterday. They are in the develop branch. I'm not opposed to the idea of merging with develop and some people test before merge to master.

TheAndi, 2019-03-17:
I can help with testing if you want

bpohvoodoo, 2019-03-17:
Feel free!
https://github.com/FabLab-Luenen/McLighting

bpohvoodoo, 2019-03-17:
@debsahu One of the bugs i found myself.
Corrected the neopixelstuff now, too.

TheAndi, 2019-03-18:
I made a video for you, if you want to take a look:
https://www.file-upload.net/download-13542504/2019-03-18_23h56_02.mp4.html

As soon as I touch the sliders it gets laggy. But as soon as I reload, everything is fine.

Chrome Version 72.0.3626.121 (Official Build) (64-Bit)
But I remember it being the same for other browsers.

gitpower2017, 2019-03-20:
Hello bpohvoodoo,

i have testing the version from 19.03.2019.
I have the feel, this is a gut RGBW implementation.
Many thanks for the good implementation

Can you better document the feature?
What is:
- the Web-GUI the 3 Color Preset, only first have a function
- What is "Autoplay"? Hi make blinking red and green?
- for was ist "RGB-Order"
- what is "McLighting Slave Nodes"? Can I controller many McL Controller with ohne GUI, WebSocket, ...

bpohvoodoo, 2019-03-20:
- the Web-GUI the 3 Color Preset, only first have a function
try animation tri color chase or blink. with second third color set
- What is "Autoplay"? Hi make blinking red and green?
Look at the definitions!
uint32_t autoParams[][6] = {   // main_color, back_color, xtra_color, speed, mode, duration (seconds)
  {0x00ff0000, 0x0000ff00, 0x00000000, 200,  1,  5000}, // blink red/geen for 5 seconds
  {0x0000ff00, 0x000000ff, 0x00000000, 200,  3, 10000}, // wipe green/blue for 10 seconds
  {0x000000ff, 0x00ff0000, 0x00000000,  60, 14, 10000}, // dual scan blue on red for 10 seconds
  {0x000000ff, 0x00ff0000, 0x00000000,  40, 45, 15000}, // fireworks blue/red for 15 seconds
  {0x00ff0000, 0x0000ff00, 0x000000ff,  40, 54, 15000}  // tricolor chase red/green/blue for 15 seconds
};
As in original repo it is what you have defined! You can add or remove effects here! Maybe configurable by UI one day.
- for was ist "RGB-Order"
Your strip has an rgb order. normally grb for ws2812, or grbw, if you use sk6812, but there are others.
- what is "McLighting Slave Nodes"? Can I controller many McL Controller with ohne GUI, WebSocket, ...
Yes, you can! Type in the slvae ips ";" separated.

HiranChaudhuri, 2019-05-13:
I just cloned https://github.com/FabLab-Luenen/GY33_MCU.git to give it a try. After all that is the version the pull request for RGBW is pending from, right?

I got caught in surprise as support for a color sensor is required. I'll add that library although not having understood the benefit for the color sensor.

debsahu, 2019-05-13:
@HiranChaudhuri comment to disable https://github.com/FabLab-Luenen/McLighting/blob/ae219075583ff0cd8e391d75239b48c2d310317e/Arduino/McLighting/definitions.h#L23

HiranChaudhuri, 2019-05-13:
Thank you for the hint, @debsahu. I also commented out the remote control and added the libraries FastLED and NeoPixelBus_by_Makuna. However now I get a compilation error:
'
In file included from sketch/mode_custom_ws2812fx_animations.h:10:0,
                 from /home/hiran/Arduino/McLighting-RGBW/Arduino/McLighting/McLighting.ino:267:
/home/hiran/Arduino/libraries/FastLED/FastLED.h:14:21: note: #pragma message: FastLED version 3.002.006
 #    pragma message "FastLED version 3.002.006"
                     ^
In file included from /home/hiran/Arduino/libraries/FastLED/led_sysdefs.h:27:0,
                 from /home/hiran/Arduino/libraries/FastLED/FastLED.h:41,
                 from sketch/mode_custom_ws2812fx_animations.h:10,
                 from /home/hiran/Arduino/McLighting-RGBW/Arduino/McLighting/McLighting.ino:267:
/home/hiran/Arduino/libraries/FastLED/platforms/esp/8266/led_sysdefs_esp8266.h:15:17: error: conflicting declaration 'typedef uint8_t boolean'
 typedef uint8_t boolean;
                 ^
In file included from sketch/McLighting.ino.cpp:1:0:
/home/hiran/.arduino15/packages/esp8266/hardware/esp8266/2.5.1/cores/esp8266/Arduino.h:191:14: error: 'boolean' has a previous declaration as 'typedef bool boolean'
 typedef bool boolean;
              ^
In file included from /home/hiran/Arduino/libraries/FastLED/FastLED.h:65:0,
                 from sketch/mode_custom_ws2812fx_animations.h:10,
                 from /home/hiran/Arduino/McLighting-RGBW/Arduino/McLighting/McLighting.ino:267:
/home/hiran/Arduino/libraries/FastLED/fastspi.h:110:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^
'

I'm not sure: Will I have to use specific versions of the libraries? So far I installed the latest available ones.

debsahu, 2019-05-13:
untested in v2.5.1

HiranChaudhuri, 2019-05-14:
Now I tried esp8266 core 2.4.2 and the project compiles. After an upload to the ESP it seems to crash though. Here is the serial output (which repeats roughly every second):
'
Starting...Main Setup
Enabled Button Mode on PIN: 14
FS File: /config.json, size: 99B
FS File: /uistate.json, size: 104B
FS File: /stripstate.json, size: 83B
FS File: /neoconfig.json, size: 43B
FS Usage: 2259/957314 bytes
Reading config file... Opened!
Config: Parsed!
{"mqtt_host":"**********","mqtt_port":"**********","mqtt_user":"**********","mqtt_pass":**********"}

Exception (28):
epc1=0x4022a1bc epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3ffffba0 end: 3fffffd0 offset: 01a0

>>>stack>>>
3ffffd40:  3fff189c 00000000 3fff16a8 40224208  
3ffffd50:  3fff189c 00000063 3fff16a8 4021c5fd  
3ffffd60:  3fff3758 3fff3758 3ffffd80 4021c5fd  
3ffffd70:  3ffe8ff6 00000063 3fff16a8 00000001  
3ffffd80:  3fff189c 00000063 3ffffdb4 4020b73c  
3ffffd90:  00000000 00000063 3fff16a8 402101b5  
3ffffda0:  3fff18ac 3fff3554 3fff3554 3fff3758  
3ffffdb0:  3fff3798 3fff3788 3fff3758 401006a0  
3ffffdc0:  402285b0 00000000 000003e8 4010020c  
3ffffdd0:  00000000 00000000 3fff34ec 401006dc  
3ffffde0:  3ffffdb4 3ffffda4 3ffffe20 40225ce4  
3ffffdf0:  00000000 3fff34ec 3fff3494 40226177  
3ffffe00:  3fff189c 3ffffec0 3fff34ec 3fff1768  
3ffffe10:  3fff189c 3ffffec0 3fff16a8 40211696  
3ffffe20:  000e9b82 000008d3 00002000 00000100  
3ffffe30:  00000005 00000020 feefeffe feefeffe  
3ffffe40:  feefeffe feefeffe feefeffe feefeffe  
3ffffe50:  feefeffe feefeffe feefeffe feefeffe  
3ffffe60:  feefeffe feefeffe feefeffe feefeffe  
3ffffe70:  feefeffe feefeffe feefeffe feefeffe  
3ffffe80:  feefeffe feefeffe feefeffe feefeffe  
3ffffe90:  feefeffe feefeffe feefeffe feefeffe  
3ffffea0:  feefeffe feefeffe feefeffe 00000000  
3ffffeb0:  00000000 00000000 feefeffe feefeffe  
3ffffec0:  3fff3494 3fff34ec feefeffe feefeffe  
3ffffed0:  feefeffe feefeffe feefeffe feefeffe  
3ffffee0:  feefeffe feefeffe feefeffe feefeffe  
3ffffef0:  feefeffe feefeffe feefeffe feefeffe  
3fffff00:  feefeffe feefeffe feefeffe feefeffe  
3fffff10:  feefeffe feefeffe feefeffe feefeffe  
3fffff20:  feefeffe feefeffe feefeffe feefeffe  
3fffff30:  feefeffe feefeffe feefeffe feefeffe  
3fffff40:  feefeffe feefeffe feefeffe feefeffe  
3fffff50:  feefeffe feefeffe feefeffe feefeffe  
3fffff60:  feefeffe feefeffe feefeffe feefeffe  
3fffff70:  feefeffe feefeffe feefeffe feefeffe  
3fffff80:  feefeffe feefeffe feefeffe feefeffe  
3fffff90:  3fff351c feefeffe feefeffe feefeffe  
3fffffa0:  feefeffe feefeffe feefeffe 3fff1768  
3fffffb0:  3fffdad0 00000000 3fff1762 40225e4c  
3fffffc0:  feefeffe feefeffe 3ffe880c 40100739  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld
'

HiranChaudhuri, 2019-05-14:
Ok, I erased the complete flash and then uploaded the sketch. This time the captive portal came up. It seems the behaviour before was caused by stale information in the flash.

I also had to connect the strip to a different GPIO, but now I can set color and effects in the UI.
It looks like McLighting-RGBW does connect to MQTT but does not advertise itself to HomeAssistant.

HiranChaudhuri, 2019-05-14:
I want to run multiple of these devices on my network. That's why I use ENABLE_MQTT_HOSTNAME_CHIPID. It seems strings like mqtt_intopic still get populated with just the hostname.

HiranChaudhuri, 2019-05-14:
Ok, the white channel is controllable in HomeAssistant. I just had to discover how the UI renders it.
All ok now. :-)

[image]

HiranChaudhuri, 2019-05-15:
So it happened that after some time (I guess a restart of HomeAssistant was involved) the communication between McLighting-RGBW and HomeAssistant broke down. I tracked it to the wrong device name being in the advertizement message, which I fixed like this:

'''
diff --git a/Arduino/McLighting/request_handlers.h b/Arduino/McLighting/request_handlers.h
index 656ad98..ea3fe49 100644
--- a/Arduino/McLighting/request_handlers.h
+++ b/Arduino/McLighting/request_handlers.h
@@ -1326,7 +1326,7 @@ void webSocketEvent(uint8_t num, WStype_t type, uint8_t * payload, size_t lenght
             const size_t bufferSize = JSON_ARRAY_SIZE(strip->getModeCount()+ 4) + JSON_OBJECT_SIZE(11) + 1500;
             DynamicJsonDocument jsonBuffer(bufferSize);
             JsonObject root = jsonBuffer.to<JsonObject>();
-            root["name"] = HOSTNAME;
+            root["name"] = mqtt_clientid;
             #if defined(MQTT_HOME_ASSISTANT_0_87_SUPPORT)
             root["schema"] = "json";
             #else
@@ -1439,9 +1439,9 @@ void webSocketEvent(uint8_t num, WStype_t type, uint8_t * payload, size_t lenght
           const size_t bufferSize = JSON_ARRAY_SIZE(strip->getModeCount()+ 4) + JSON_OBJECT_SIZE(11) + 1500;
           DynamicJsonDocument jsonBuffer(bufferSize);
           JsonObject root = jsonBuffer.to<JsonObject>();
-          root["name"] = HOSTNAME;
+          root["name"] = mqtt_clientid;
           #if defined(MQTT_HOME_ASSISTANT_0_87_SUPPORT)
-          root["schema"] = "json";
+            root["schema"] = "json";
           #else
           root["platform"] = "mqtt_json";
           #endif
'''