toblum/McLighting #362: Unable to compile with ENABLE_MQTT enabled

macadamer, 2019-03-16:
As subject, with MQTT enabled sketch (version 2.2.2) will not compile. My knowledge of Arduino/C programming is insufficient to unpick this error.

Arduino: 1.8.9 Hourly Build 2019/01/03 05:21 (Mac OS X), Board: "LOLIN(WEMOS) D1 mini Pro, 80 MHz, Flash, Disabled, 16M (14M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

In file included from /Users/derryck/Documents/Arduino/McLighting/Arduino/McLighting/McLighting.ino:253:0:
sketch/request_handlers.h: In function 'void mqtt_reconnect()':
request_handlers.h:999:84: error: request for member 'c_str' in 'mqtt_will_topic', which is of non-class type 'const char [20]'
       if (mqtt_client.connect(mqtt_clientid, mqtt_user, mqtt_pass, mqtt_will_topic.c_str(), 2, true, mqtt_will_payload, true)) {
                                                                                    ^
exit status 1
request for member 'c_str' in 'mqtt_will_topic', which is of non-class type 'const char [20]'

I have another problem with AMQTT enabled which leads to constant Wemos reboots, but let's get one thing working at a time :-)

debsahu, 2019-03-16:
A bug https://github.com/toblum/McLighting/issues/340

.c_str() needs to be removed see d6297ae9543b32d60f1a66cd4eafae0220a32fb8

macadamer, 2019-03-16:
Ah - sorry, that should have been obvious! Compiles without errors now, many thanks.

closed 2019-03-16 by macadamer