toblum/McLighting #365: Save the last state?

moddok, 2019-03-20:
Hey,

is it somehow possible to save the last state/settings of the LEDs? everytime i power off the arduino (Wemos D1 Mini -> ESP8266) and power it back on again, the LEDs are light up in static red. no settings are saved.
is there something in the code i can change, that the wemos will save the last state?

thank you in advance for your answers!

debsahu, 2019-03-20:
Last state is stored in SPIFFs, make sure you have correct layout chosen when flashing: 4MB flash, 3MB SPIFFs.

moddok, 2019-03-20:
thank you! works perfect 💯 👍 
but i have another question: is it somehow possible to deactivate the onboard LED?

bpohvoodoo, 2019-03-21:
Yes, search for digitalWrite(LED_BUILTIN, LOW) or  digitalWrite(BUILTIN_LED, LOW); in McLighting.ino
and change LOW to HIGH

closed 2019-03-29 by debsahu