toblum/McLighting #366: Information transfer via mqtt

Teckware, 2019-03-20:
I'd like to attach your sketch to the voltage to measure the voltage when I run my circuit through a battery-backed solar panel. In the serial monitor I get it already displayed but I do not get it via mqtt transfer.
############################
{
 float temp;
 val=analogRead(0);
 temp=val/4.092;
 val=(int)temp;//
 //val1=((val%100)/10);
 delay(60000);
 Serial.println(String(val).c_str()); 
 client.publish(voltageBattery_topic, String(val).c_str(), true);
 }
##############################

Did you have an idea.???

Best regards,
Teckware

debsahu, 2019-03-20:
What MQTT are you using: PubSubClient or AsyncMQTT? Don't use delays!

Teckware, 2019-03-21:
Hi Depsahu, sorry for the delay. I use AMQTT and I know that this statment is wrong. Normaly it is uncoment. But for this post i have made only copy paste. I think this :

client.publish(voltageBattery_topic, String(val).c_str(), true); 

is the wrong sektion. When i use the top sektion without the client.public piece i hafe all 60 sec a comment in my serial window but i would like to send it out with AMQTT. Preferably to the State comment from your sketch. but with this:

voltageBattery_topic

I have generated a privat MQTT string. (I think so)

Best regards,
Teck

debsahu, 2019-03-29:
Please use this as [reference for async MQTT](https://github.com/marvinroger/async-mqtt-client/blob/master/docs/2.-API-reference.md#uint16_t-publishconst-char-topic-uint8_t-qos-bool-retain-const-char-payload--nullptr-size_t-length--0-bool-dup--false-uint16_t-message_id--0)

Correct way to do this for AMQTT would be:
`client.publish(voltageBattery_topic, 0, true, String(val).c_str());`

closed 2019-03-29 by debsahu