toblum/McLighting #368: Help Please - Soft WDT reset constantly

lanky8804, 2019-03-28:
Installed and flashed all ok but log produces this and constantly restarts. Cannot get to IP address due to this, any help appreciated.

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld
FS File: /config.json, size: 61B
FS File: /neoconfig.json, size: 43B
FS Usage: 1004/957314 bytes

Reading neoconfig file... Opened!
Config:  Parsed!
{"pixel_pount":300,"rgb_order":82,"pin":15}Reading config file... Opened!
Config:  Parsed!
{"mqtt_host":"","mqtt_port":"","mqtt_user":"","mqtt_pass":""}WiFiManager config FS Read success!
*WM: Adding parameter
*WM: host
*WM: Adding parameter
*WM: port
*WM: Adding parameter
*WM: user
*WM: Adding parameter
*WM: pass
*WM: Adding parameter
*WM: strip_size
*WM: Adding parameter
*WM: led_pin
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
Saving Strip cfg: {"pixel_pount":300,"rgb_order":82,"pin":15}
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.1.125
SaveConfig is False!
WiFiManager config FS Save failure!
connected...yeey :)
Open http://192.168.1.125/ to open McLighting.
Use http://McLighting01.local/ when you have Bonjour installed.
New users: Open http://192.168.1.125/upload to upload the webpages first.

E1.31 mode setup complete.
Couldn't find "/stripstate.json"
 Failure!
*WM: freeing allocated params!
SET_MODE: 0 0

Soft WDT reset

ctx: cont 
sp: 3fff2880 end: 3fff2af0 offset: 01b0

>>>stack>>>
3fff2a30:  3ffe8e4d 3ffe9ba9 000003e8 c018f064  
3fff2a40:  00000000 000000ff 00000000 4022cdef  
3fff2a50:  3fff123c 000000cc 402242c0 00000210  
3fff2a60:  3fff5234 3fff5034 0000ae7f 4021f748  
3fff2a70:  3ffe94a5 3fff0e08 00000000 000014d9  
3fff2a80:  3fff5234 00000000 3fff5034 4021f3f8  
3fff2a90:  ffffff80 3fff0de4 3fff5034 4021f36d  
3fff2aa0:  3fff5034 3fff0de4 3fff5034 3fff18c0  
3fff2ab0:  3fff5034 3fff0de4 3ffe8550 4020fd41  
3fff2ac0:  feefeffe feefeffe feefeffe 3fff1ac4  
3fff2ad0:  3fffdad0 00000000 3fff1abe 4022430c  
3fff2ae0:  feefeffe feefeffe 3fff1ad0 40100710  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

tooljose88, 2019-03-29:
I have the same issue

debsahu, 2019-03-29:
Could you decode the stack? https://github.com/me-no-dev/EspExceptionDecoder

lanky8804, 2019-03-29:
decoder and got this, makes very little sense to me im afraid.

Decoding stack results
0x4022cdef: exp at ../../../.././newlib/libm/math/w_exp.c line 89
0x402242c0: esp_yield() at /Users/.../Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/core_esp8266_main.cpp line 87
0x4021f748: WS2812FX::mode_static() at /Users/.../Documents/Arduino/libraries/WS2812FX-master/src/WS2812FX.cpp line 527
0x4021f3f8: WS2812FX::service() at /Users/.../Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algobase.h line 222
0x4021f36d: WS2812FX::show() at /Users/.../Documents/Arduino/libraries/WS2812FX-master/src/WS2812FX.cpp line 131
0x4020fd41: loop() at /Users/.../Documents/McLighting-master/Arduino/McLighting/McLighting.ino line 1295
0x4022430c: loop_wrapper() at /Users/.../Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/cores/esp8266/core_esp8266_main.cpp line 121

debsahu, 2019-03-29:
Firstly recommended core is 2.4.2 and recently we have tested with 2.5.0. It looks like you are using core 2.4.1.

The stack results are pointing to some math calculations form WS2812FX is causing the wdt reset. I would update all libraries mentioned [here](https://github.com/toblum/McLighting#used-libraries)  to latest version.

lanky8804, 2019-03-30:
Thank you for the support, updated to 2.5.0 now for core and updated libraries to latest versions. Just to double check some of these libraries have newer versions than stated on the webpage i have just gone for the newest of each.

I am still getting this error.

Decoding stack results
0x40222a2c: WS2812FX::mode_static() at /Users/../Documents/Arduino/libraries/WS2812FX-master/src/WS2812FX.cpp line 527
0x402226dc: WS2812FX::service() at /Users/../Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algobase.h line 222
0x40222651: WS2812FX::show() at /Users/../Documents/Arduino/libraries/WS2812FX-master/src/WS2812FX.cpp line 131
0x4021033d: loop() at /Users/../Documents/McLighting-master/Arduino/McLighting/McLighting.ino line 1295
0x40227a80: loop_wrapper() at /Users/../Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_main.cpp line 125

some screenshots from WS2812 + code lines
[image]

[image]

lanky8804, 2019-03-30:
Whats the best version to use for WS2812FX as I'm on 1.1.8

debsahu, 2019-03-30:
I have another method for you to try, using precompiled binaries. Give me a few days, I'm in the process of making a video on this.

debsahu, 2019-03-30:
You can use the latest version of all libraries. The certain of libraries in the repo are the minimum version to get everything to work properly. I'm surprised that many people are getting a reset in the loop() or ws2812fx's service()

gulliverrr, 2019-04-02:
Same here on a NodeMCU :(

gulliverrr, 2019-04-02:
Some extra info trying to troubleshoot this:

- If I don't enable any MQTT (either ENABLE_AMQTT or ENABLE_MQTT) works fine from a browser using the buttons on the page
- If I enable either ENABLE_AMQTT or ENABLE_MQTT soft resets occur

lanky8804, 2019-04-02:
Tried on 2 different nodemcu and a wemos d1. Still same issue.

gulliverrr, 2019-04-03:
Throwing the same stack:

    0x4021c8f4: WS2812FX::mode_static() at /home/user/.arduino15/libraries/WS2812FX/src/WS2812FX.cpp line 527
    0x4021c5a4: WS2812FX::service() at /home/user/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algobase.h line 222
    0x4021c519: WS2812FX::show() at /home/user/.arduino15/libraries/WS2812FX/src/WS2812FX.cpp line 131
    0x4020e3e9: loop() at /tmp/arduino_modified_sketch_501303/McLighting.ino line 1293
    0x40221110: loop_wrapper() at /home/user/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_main.cpp line 125`

McLighting.ino line 1293 is the if (updateStateFS):

      #ifdef ENABLE_STATE_SAVE_SPIFFS
        if (updateStateFS) {
          (writeStateFS()) ? DBG_OUTPUT_PORT.println(" Success!") : DBG_OUTPUT_PORT.println(" Failure!");
        }
      #endif

WS2812FX.cpp line 131 is the Adafruit_NeoPixel::show(); :  

    void WS2812FX::show(void) {
      if(customShow == NULL) {
        Adafruit_NeoPixel::show();
      } else {
        customShow();
      }
    }

I believe it has something to do with the filesystem, writing where is not supposed to as this is the major difference between using MQTT and not. Having to store somewhere the values. Trying to make the 4 fields smaller (host, port, user, pass) didn't hide the error.

debsahu, 2019-04-03:
Could you try the develop version? Made a check if strip is initialized before service() is called.

gulliverrr, 2019-04-03:
Tried PlatformIO too in case Arduino IDE was causing something in the compilation of the libraries (long shot, I know) but of course the exact same issue with the reset continues... 
Your help will be very much appreciated.

gulliverrr, 2019-04-03:
:(
same reset:

```
0x4021f3ac: WS2812FX::mode_static() at /home/user/.arduino15/libraries/WS2812FX/src/WS2812FX.cpp line 527
0x4021f05c: WS2812FX::service() at /home/user/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algobase.h line 222
0x4021efd1: WS2812FX::show() at /home/user/.arduino15/libraries/WS2812FX/src/WS2812FX.cpp line 131
0x4020e740: loop() at /tmp/arduino_modified_sketch_139003/McLighting.ino line 1301
0x40223bc8: loop_wrapper() at /home/user/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_main.cpp line 125
```

dale3h, 2019-04-03:
I am having what I believe is the same issue on a Wemos D1 Mini. I have discovered by trial and error that the last commit (to master) that is working for me is [`LWT for MQTT and AMQTT`](https://github.com/toblum/McLighting/commit/38d8f3f8134837b1d0ece6a192a51bc9ad63ff2a).

As soon as I try to flash [`bug_fix`](https://github.com/toblum/McLighting/commit/8ff869ce7fb5d1688694b6c6944986fb65fcadc0) I start seeing the reboot loop.

I am also trying the latest in `develop` branch and will report back shortly.

dale3h, 2019-04-03:
Same result for me using latest code in `develop`.

**@gulliverrr @lanky8804 @tooljose88 and anyone else that is experiencing this same issue,** could you please try flashing [`Add LWT for MQTT and AMQTT`](https://github.com/toblum/McLighting/commit/38d8f3f8134837b1d0ece6a192a51bc9ad63ff2a) and see if the exceptions and reboot loop disappear?

If they do, please then try flashing [`bug_fix `](https://github.com/toblum/McLighting/commit/8ff869ce7fb5d1688694b6c6944986fb65fcadc0) to see if they come back.

Please report back your findings so that we can pin-point where this bug was introduced.

gulliverrr, 2019-04-03:
[Add LWT for MQTT and AMQTT ](https://github.com/toblum/McLighting/commit/38d8f3f8134837b1d0ece6a192a51bc9ad63ff2a) does work indeed. Testing bug_fix now...

dale3h, 2019-04-03:
Perfect, thank you so much for trying that out. Now we have two solid confirmations that it is something in [`bug_fix`](https://github.com/toblum/McLighting/commit/8ff869ce7fb5d1688694b6c6944986fb65fcadc0) that breaks it on our boards.

gulliverrr, 2019-04-03:
All those `.c_str()` gone... ?

dale3h, 2019-04-03:
I am playing with a few things as well, but I don't think it's the change of removing `.c_str()`. I have not confirmed this yet, but the fact that execution passes those successfully before it throws an exception leads me to think it is not something in `setup()`, but rather `loop()`.

dale3h, 2019-04-03:
Okay, I think I'm making some sort of progress, but I still don't have the answer. Assuming you're still on [`bug_fix`](https://github.com/toblum/McLighting/commit/8ff869ce7fb5d1688694b6c6944986fb65fcadc0), if you comment out [line 1249 of `McLighting.ino`](https://github.com/toblum/McLighting/blob/8ff869ce7fb5d1688694b6c6944986fb65fcadc0/Arduino/McLighting/McLighting.ino#L1249) (`strip->service();`), the reboot loop disappears. This change results in the strip being completely disabled, though.

However, from the diff it appears that this line remained unchanged on this commit, so I don't know.

debsahu, 2019-04-03:
Thanks for pointing these out, a few things happened in that commit.
- changed Strings to char arrays, hence no need to convert Strings to .c_str()
- added custom values to WiFi Manager to initialize the strip with values entered by user during initial setup
-REST API was added to change LED settings.

I'll investigate. Cant reproduce on my NodeMCU for some reason.

debsahu, 2019-04-03:
Comment out these lines in **McLighting.ino**
```
    strcpy(strip_size, custom_strip_size.getValue());
    strcpy(led_pin, custom_led_pin.getValue());

    if(atoi(strip_size) != WS2812FXStripSettings.stripSize)
      WS2812FXStripSettings.stripSize = atoi(strip_size); 
    uint8_t pin = atoi(led_pin);
    if ((pin == 16 or pin == 5 or pin == 4 or pin == 0 or pin == 2 or pin == 14 or pin == 12 or pin == 13 or pin == 15 or pin == 3 or pin == 1) and (pin != WS2812FXStripSettings.pin) )
      WS2812FXStripSettings.pin = pin;    
    initStrip();
```

dale3h, 2019-04-03:
Just to confirm, I am still on [`bug_fix`](https://github.com/toblum/McLighting/commit/8ff869ce7fb5d1688694b6c6944986fb65fcadc0). After commenting out the specified lines, here are the results:

Stack from first reset:
```
0x40100216: millis at /Users/Dale/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_wiring.c line 183
0x40222558: WS2812FX::setPixelColor(unsigned short, unsigned char, unsigned char, unsigned char) at /Users/Dale/Documents/Arduino/libraries/WS2812FX/src/WS2812FX.cpp line 105
0x4021028d: loop() at /Users/Dale/Projects/McLighting/Arduino/McLighting/McLighting.ino line 1243
0x402278dc: String::trim() at /Users/Dale/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/WString.cpp line 778
```

Stack from second reset:
```
0x4022790e: String::toInt() const at /Users/Dale/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/WString.cpp line 788
0x40223254: WS2812FX::scan(unsigned int, unsigned int, bool) at /Users/Dale/Documents/Arduino/libraries/WS2812FX/src/WS2812FX.cpp line 765
0x40100216: millis at /Users/Dale/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_wiring.c line 183
0x40222558: WS2812FX::setPixelColor(unsigned short, unsigned char, unsigned char, unsigned char) at /Users/Dale/Documents/Arduino/libraries/WS2812FX/src/WS2812FX.cpp line 105
0x4021028d: loop() at /Users/Dale/Projects/McLighting/Arduino/McLighting/McLighting.ino line 1243
0x402278dc: String::trim() at /Users/Dale/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/WString.cpp line 778
```

Stack from third reset is same as second.
Stack from fourth reset is same as first.

dale3h, 2019-04-03:
I attempted to flash latest [`develop`](https://github.com/toblum/McLighting/commit/cc94b8de93fb2bfb7df79c8e27b27c8539c09e54) to see if that helped, but it does not compile:

```
/Users/Dale/Projects/McLighting/Arduino/McLighting/McLighting.ino: In function 'void setup()':
McLighting:381:11: error: 'strip_size' was not declared in this scope
   sprintf(strip_size, "%d", WS2812FXStripSettings.stripSize);
           ^
McLighting:382:11: error: 'led_pin' was not declared in this scope
   sprintf(led_pin, "%d", WS2812FXStripSettings.pin);
           ^
exit status 1
'strip_size' was not declared in this scope
```

debsahu, 2019-04-03:
sorry, made local changes that did not get pushed. It should compile now. https://github.com/toblum/McLighting/commit/b459b240149da1431e847b56f4b6bd62e3a9e2ee

dale3h, 2019-04-03:
It compiles, but still no dice on Wemos D1 Mini. :(

```
0x4022791e: __yield() at /Users/Dale/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_main.cpp line 100
0x40223264: WS2812FX::mode_rainbow_cycle() at /Users/Dale/Documents/Arduino/libraries/WS2812FX/src/WS2812FX.cpp line 812
0x40100216: millis at /Users/Dale/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_wiring.c line 183
0x40222568: WS2812FX::service() at /Users/Dale/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algobase.h line 222
0x402103b0: loop() at /Users/Dale/Projects/McLighting/Arduino/McLighting/McLighting.ino line 1302
0x402278ec: loop_wrapper() at /Users/Dale/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_main.cpp line 125
```

gulliverrr, 2019-04-03:
Well, I have "fixed" master but not sure how... :)
request_handlers.h has 2 occurrences of this typo
`json["pixel_pount"]` instead of `json["pixel_count"]` but I think if they match it is no problem.

Then if I hardcode WS2812FXStripSettings.stripSize commenting out these lines
```
    if(atoi(strip_size) != WS2812FXStripSettings.stripSize)
      WS2812FXStripSettings.stripSize = atoi(strip_size); 
```

it works...

debsahu, 2019-04-03:
omg, how did I miss that!

gulliverrr, 2019-04-03:
Oh, line 899 in McLighting.ino is actually looking for `pixel_count`
    json["pixel_count"] = WS2812FXStripSettings.stripSize;

I think we got it... ;)

lanky8804, 2019-04-04:
Develop version compiled and uploaded to a NoduMCU and working fine. Master still not working but might not include fixes above as yet?

debsahu, 2019-04-04:
Once the develop solves all issues here, we will merge with master.

lanky8804, 2019-04-04:
@debsahu  ok thank you. Random question when you define the LED_PIN does it have to be like '11, 14...'  or can you define it D1 ...

dale3h, 2019-04-04:
@lanky8804 Because `LED_PIN` uses the `#define` language construct (which Arduino replaces for you before compiling), you can use any value that your underlying libraries/boards declare/define, as long as the type matches and/or is compatible. Because McLighting expects `LED_PIN` to be of type `uint8_t`, and Arduino pin definitions for the boards we are using are of the same type `uint8_t`, it should work without any issues. Arduino IDE will quickly let you know if it's not, as it will not compile successfully.

For example, if I select **NodeMCU** as the board to which I am flashing the sketch, Arduino uses [this file](https://github.com/esp8266/Arduino/blob/2.5.0/variants/nodemcu/pins_arduino.h) for the pin definitions. For the **D1 Mini** it uses [this file](https://github.com/esp8266/Arduino/blob/2.5.0/variants/d1_mini/pins_arduino.h), which is mostly the same except for a couple of small differences.

**tl;dr:** At time of post, you can use both formats (`14` or `D5`) for `LED_PIN` in **McLighting**.

dale3h, 2019-04-04:
Okay, I was able to get latest `develop` to compile ***and*** work on Wemos D1 Mini with hardcoding the pin. I will play around with it to see if I can get it working as expected, and if so, I'll submit a PR against `develop`.

dale3h, 2019-04-04:
I think I have found one more clue:
```
sprintf(strip_size, "%d", WS2812FXStripSettings.stripSize);
```
For some reason this doesn't print the integer value of `WS2812FXStripSettings.stripSize` to `strip_size`. When I output `strip_size` after the `sprintf`, it's just blank.

dale3h, 2019-04-04:
***I FINALLY FIGURED IT OUT!!!***

In Arduino and C++, what's the first thing you do when you expect a char array to have a value, but the output of the char array is blank? I don't know about you, but my go-to is to iterate the char array and output the hex values of each char in the array!

Using this tactic, I ~~quickly~~ realized that the char array declarations for `strip_size` and `led_pin` do not include room for the terminating null character. When I iterated over `strip_size` (value = `24`) it spit out `0x00 0x34 0x00`, which is incorrect -- it *should* have been `0x32 0x34 0x00`. I did the same for `led_pin` (value = `14`) just for comparison and the output was `0x31 0x34`, which at first glance is correct, but it's missing the terminating null character. I guess I just got lucky and whatever happened to be at the address immediately following `led_pin` started with a null character.

Now, why `sprintf` was working for `led_pin` but not for `strip_size`, I have no freakin' clue! I had the hypothesis that it was because `led_pin` was using its full allocated length of 2 characters (`0x31 (1)` and `0x34 (4)`, but `strip_size` was only use 2 of its 3 (`0x32 (2)` and `0x34 (4)`). However, I tried `WS2812FXStripSettings.stripSize = 124;` (leaving the char array lengths unchanged) and it resulted in `0x00 0x32 0x34`.

I would apologize for the programming lecture, but I have hope that someone some day will absorb some amount of knowledge from my ramblings.

**tl;dr:** The fix is super simple. Aside from changing the typos pointed out by @gulliverrr (`pixel_pount` -> `pixel_count`), the declared lengths of the char arrays in `request_handlers.h` need to be changed to `4` and `3` respectively:

```
char strip_size[4], led_pin[3]; //needed for WiFiManager Settings
```

I am going to test the changes against a handful of recent commits and see if that's all it takes. I have a feeling that because of the typos, we might also have to check `pixel_count` for a positive value, and set it to the default if not.

debsahu, 2019-04-04:
What I don't understand is I removed all these code from the develop version and still there are issues.

dale3h, 2019-04-04:
I tested against the following and each works after adding the changes:

- [`d6be205`](https://github.com/toblum/McLighting/commit/d6be20546ade208e6e3f8b4c05320ac91a90c195`)
- [`a9a3565`](https://github.com/toblum/McLighting/commit/a9a35651458d36148365cad056764c147f446d23)
- [`0fee256` (current master at time of comment)](https://github.com/toblum/McLighting/commit/0fee256c3abd5b7e6c40cccf05f35025a7124555)
- [`bug_fix` (commit that initially broke)](https://github.com/toblum/McLighting/commit/8ff869ce7fb5d1688694b6c6944986fb65fcadc0) - this one has an extra typo in `McLighting.ino` on line 852, but it only affects the output of the deprecated `/pixels` API endpoint.

The changes (all in `request_handlers.h`):

1.
```
char strip_size[3], led_pin[2]; //needed for WiFiManager Settings
```
change to:
```
char strip_size[4], led_pin[3]; //needed for WiFiManager Settings
```

2.
```
WS2812FXStripSettings.stripSize = json["pixel_pount"];
```
change to:
```
WS2812FXStripSettings.stripSize = (json["pixel_count"]) ? json["pixel_count"] : NUMLEDS;
```

3.
```
json["pixel_pount"] = WS2812FXStripSettings.stripSize;
```
change to:
```
json["pixel_count"] = WS2812FXStripSettings.stripSize;
```

dale3h, 2019-04-04:
@debsahu That is a very good point. I am taking a look back at `develop` to see if the issue is related.

dale3h, 2019-04-04:
I am getting the PR together right now.

gulliverrr, 2019-04-05:
I can confirm the char array size change saved the day. Applied the pount to count and the move of the #ifdef from your commit and flashed NodeMCU with Arduino IDE with these settings:
[image]
Oh and hard coded these in definitions.h
```
  char mqtt_host[64] = "10.10.10.150";
  char mqtt_port[6] = "1883";
```
to have the fields ready (but editable if I change my setup) just to save me from typing (or remembering 1883).

> When I iterated over `strip_size` (value = `24`) it spit out `0x00 0x34 0x00`, which is incorrect -- it _should_ have been `0x32 0x34 0x00`. I did the same for `led_pin` (value = `14`) just for [...]

I believe the `led_pin` is allocated memory first and when a 3 char (including terminator) variable is assigned to it, the terminator slips to the memory adjacent which happens to be of `strip_size`, hence the 0x00 at the beginning. Unluckily assigning the values of these 2 vars is always happening first for the `strip_size` and then for the `led_pin`. 
Incorrect definition and initialisation of vars causes this silent bugs lurking for the right moment to hit (or a great combination of circumstances)!
Great work everybody!
Now I can go on my ladder and install those WS's !

commit a9a35651 (2019-04-03): fix #368

- see if this fixes https://github.com/toblum/McLighting/issues/368
closed 2019-04-16 by debsahu