toblum/McLighting #369: wemos d1 mini pro v2

alfredolo, 2019-03-30:
hello.
I have a wemos d1 mini pro v2 and some ws2815 12v led strip.
When i connect all i have some issue :
- some led strips work great
- some led strips have problems. Led random flickers or sometimes they have the correct colors  but only half of them. 1 led is correct and 1 led dont.
If i disconnect wemos d1 mini pro v2 and I connect wemos mini pro v1.1 it work like aspected.
So i think there is a little difference between the 2 boards.
Any idea how to fix it ?
Thank you

HiranChaudhuri, 2019-05-07:
Did you check the voltages for power and signal? AFAIK the D1 mini pro operates at 3.3 while your LED strip seems to use 12 V. So what about the signals you transmit?

alfredolo, 2019-05-09:
Thank you for reply.
You right, i fixed the problem with a level shifter from 3.3 to 12v.
The real problem is : why wemos d1 mini pro 1.1 work with 3.3v output and wemod d1 mini pro v2 dont ?

HiranChaudhuri, 2019-05-09:
I'm afraid I do not understand. The setup you had should have never worked, and with a level shifter it did. If you say it worked before, you may have been lucky. I have no explanation.

closed 2019-07-05 by toblum