toblum/McLighting #370: Multiple commands via MQTT ignored

nicr4wks, 2019-04-03:
Hi Team,

Not 100% certain if this one is a bug or intended behavior.

I've noticed when sending multiple commands via MQTT, RGB and brightness at the same time for example. `{"state": "ON", "color": {"r": 0, "g": 255, "b": 255}, "brightness": 100}` only the brightness command is executed and the color stays the same. I can resubmit this command as many times as I like the color will not change.

If I submit the RGB portion on its own the color changes fine.

I've worked around this in my  automations by submitting RGB and brightness one after the other but wanted to check what the official word on this was.

Cheers.

debsahu, 2019-04-03:
Wiki has details on how to use the MQTT API, which is same as websocket API. What you are referring to is home assistant spending stuff, HA sends ON, RGB and brightness separately. Hence the behavior you are seeing.

ryansabin, 2019-08-08:
If you want to send all the settings in one command you can use the HA MQTT command. But, it always sets the color via color_temp using that command. It actually sets the color using RGB then goes on and resets the color using color_temp. So you can figure out how to convert RGB to mireds and use the HA MQTT command.

Or, as  I do so I can send 1 command to update all settings. I commented out the part where it sets the color using color_temp. You still include the color_temp. Mclighting just ignores it.

In version 2.2 you need to comment out line 913-919 in request_handlers.h. I always just search for color_temp in that file and comment the following section:

//      if (root.containsKey("color_temp")) {
//        //temp comes in as mireds, need to convert to kelvin then to RGB
//        color_temp = (uint16_t) root["color_temp"];
//        unsigned int kelvin  = 1000000 / color_temp;
//        main_color = temp2rgb(kelvin);
//        mode = SETCOLOR;
//      }

If you are using HA to control your lights THIS WILL BREAK IT!