toblum/McLighting #371: e131 mode switching bug

colarocker, 2019-04-03:
When i switch mode to e131 using the webinterface and the button, it works without a problem
but using the http get with http://hostname/e131, the LEDs stay in the previous selected mode, like for example static.
As soon as an e131 signal is sent with a software like jinx, it starts to switch constantly between the e131 signal and the selected mode through the webinterface before.

debsahu, 2019-04-03:
Thanks for reporting this, it is indeed a bug.

You can add 

```
if(strip->isRunning()) strip->stop();
```

before this line to fix this https://github.com/toblum/McLighting/blob/0fee256c3abd5b7e6c40cccf05f35025a7124555/Arduino/McLighting/McLighting.ino#L1063

colarocker, 2019-04-03:
Thanks for the fast fix! Really appreciate your work!

closed 2019-05-01 by debsahu