toblum/McLighting #379: Wrong Colors

NicoEich, 2019-04-15:
Hey guys, i am not sure if this is a software or a hardware problem but i use this software on an esp2866-01(1MB) installed with PlatformIO to controll some WS2812B (https://www.amazon.de/gp/product/B07KGB5FR6/ref=ppx_yo_dt_b_asin_title_o04_s00?ie=UTF8&psc=1).

And i am getting the wrong colors, the are pretty random and i cant see an order. I tried it with "#define USE_WS2812FX_UART1" and without it on LED_PIN 2, with two different esp modules. Now i am bit stuck here and would appreciate any kind of help.

[image]

Here is my test setup, the leds are getting 5v/20A from a powersupply and the esp is powerd by an usb adapter. I tried it with and without the 220Ω resistor it shows the same issue.

toblum, 2019-04-17:
Hi @NicoEich,
if you're using the latest version, you can change the color scheme via HTTP call. Have a look here: https://github.com/toblum/McLighting/issues/367

Regards
Tobias

NicoEich, 2019-04-17:
Hi, thanks for the fast reply. Is the default color scheme not GRB ?

Did some more research today and maybe my data ouput voltage could be to low. The WS2812b datasheet says that the input data voltage should be 70% of the VCC, so with 5V VCC it should be at least 3,5V data voltage. The esp-01 has just 3.3V so i have to test this out first.

Has anyone here used an Esp 2866-01 to control WS2812b and could show me his data connection ?

debsahu, 2019-04-17:
Do you have **common ground**?

Assuming `#define USE_WS2812FX_UART1`

Power      -> WS2812B
--------------------------
5V Power -> VIN
GND        -> GND
--------------------------

ESP8266   -> WS2812B
--------------------------
GPIO2   -> 220Ω resistor (optional) -> DATA 
GND     -> GND
--------------------------

USB ADAPTER -> ESP8266
------------------------------
5V     ->  5V LDO
GND -> GND

NicoEich, 2019-04-17:
Thanks, no they dont have a common ground. I thought the power ground would be enough.
So i have to connect the ESP GND with the WS2812B GND and the USB Ground, gonne do that tomorrow.

NicoEich, 2019-04-20:
So i can finally say its working, i had to sacrifi one led as power shifter and connect the led ground with esp ground. The sacrificial led is explained here:
 ( https://hackaday.com/2017/01/20/cheating-at-5v-ws2812-control-to-use-a-3-3v-data-line/ )

Thank you for your help

closed 2019-04-20 by NicoEich