toblum/McLighting #380: are the APA102 also supported?

salopette, 2019-04-15:
Thank you for the great project!
I have a question, are the APA102 Leds also supported?
If so, what do I have to change in the config?

Thanks again!

toblum, 2019-04-17:
HI @salopette,

I'm not sure as I don't own these. 
I'm quite sure that they don't work for standard setups as the Adafruit Neopixel Lib doesn't support them. It might work for setups that use the NeoPixelBus Library (USE_WS2812FX_DMA, USE_WS2812FX_UART) but that's just a guess.
If you could try it out, please drop us a note if it worked.

Regards
Tobias

closed 2019-04-17 by toblum

debsahu, 2019-04-17:
NeoPixelBus uses DotStarMethod for APA102 on SPI bus with support for both hardware and software SPI methods. Not yet implemented here!

toblum, 2019-04-17:
I wasn't sure here, thanks for the clarification!

salopette, 2019-04-17:
so it does not work?

salopette, 2019-04-18:
https://github.com/Makuna/NeoPixelBus

APA102 - SPI based chipset

debsahu, 2019-04-18:
I tried and it doesn't work directly. Dotstar/APA102 needs sending of brightness data along with RGB for every LED. I have a different solution, be please be patient until it gets imported here.

salopette, 2019-04-19:
that would be fantastic, I'm already waiting impatiently!

craigsincl, 2019-08-14:
Hello, is this still in the pipeline? Thank you for all your hard work on this project.