toblum/McLighting #382: Problem with the LED number change

3D-Drucker-Kram, 2019-04-16:
Hello, tobias,
I loaded the MClighting 2.2.3 from you. There is probably a problem there. The LED's works BUT after I've counted the LED chain (57) and wanted to change the birth value of 24 to 57 LEDs, only the 24 LEDs are activated. It can not be changed or it has no effect. Have downloaded version 2.2.3 down again and re-uploaded but no effect. The number of controllable LED's I can not change. If I try it with the 2.2.0 then he takes the 57 LED's.

Please help
Unfortunately, I find no other side to reach you directly.
Greeting Ralf

Hallo Tobias,
ich hab da die MClighting 2.2.3 von dir geladen. Es gibt da wohl möglich ein Problem. Das mit den LED´s klappt ABER nachdem ich jetzt mal die LED-Kette gezählt habe (57) und den Geburtswert von 24 ändern wollte auf 57 LED´s werden immer nur die 24 LED´s angesteuert. Es kann nicht geändert werden, bzw. es hat keine Auswirkung. Hab die Version 2.2.3 nochmal runter geladen und neu hochgespielt aber ohne Wirkung. Die anzahl der ansteuerbaren LED´s kann ich nicht ändern. Wenn ich es mit der 2.2.0 versuche dann nimmt er die 57 LED´s.

Bitte um Hilfe
Leider finde ich keine andere Seite dich direkt zu erreichen.
Gruß Ralf

debsahu, 2019-04-16:
try 2.2.4

debsahu, 2019-04-17:
Please erase flash completely and use rest API https://github.com/toblum/McLighting/wiki/HTTP-REST-API#get-pixelconf-http10 to change the number of pixels

3D-Drucker-Kram, 2019-04-17:
OK Thanks for the answer, but I do not understand. Unfortunately, I am only a user and not a programmer. Can you explain that to me?

toblum, 2019-04-17:
Hi @3D-Drucker-Kram,

if you'Reusing the Arduino IDE, you find a setting "Erase flash" under "tools". Set it to "all contents" and reflash the board.
To change the number of LEDs just follow the link provided by @debsahu for more informations. If McL is running just call http://<mc_lighting_ip_or_hostname>/pixelconf?ct=57 to set the number of LEDs to 57.

Greetings
Tobias

3D-Drucker-Kram, 2019-04-19:
Great, that was the solution! Thank you for your prompt reply.

toblum, 2019-04-19:
Great!

closed 2019-04-19 by toblum