toblum/McLighting #386: different speed

GrumpyNyanCat, 2019-04-19:
Hello,

I'm running 240 WS2812b @ nodemcu and 120 WS2813 @ wemos d1 mini (both should have the esp8266-12e)

Everything in McLighting is default, except that I hat to enable "define USE_WS2812FX_DMA" on the wemos to make the WS2813 work. 

Now my problem ist, that the FX speed of both setups are different from each other. 
For example "Rainbow(11)" is changing the color on the nodemcu(2812b) at 220 speed approximately once every second, but on the wemos(2813) the colors are going crazy, like three times faster. 

It's like this with every animation I tested. 

What's happening here?
Is it because of the "define USE_WS2812FX_DMA" ?

debsahu, 2019-04-19:
The animation speed is related to the number of LEDs you have and not the method used or type of LED. This project is based on WS2812FX which has all the animations coded in.

GrumpyNyanCat, 2019-04-19:
ah okay, thank you very much. haha, now I feel a bit stupid. So I have to dive deeper into WS2812FX code to fix this for me.

closed 2019-05-01 by debsahu