toblum/McLighting #388: Change start led

Maurotb, 2019-04-21:
Hi,
i have 894 led ws2815 ,
If i connect my d1 mini pro af first led , i have 6mt of cable and effect not work correctly.
If i connect my board at led  ~100 , i have 1mt of cable and all work correctly but some
effect start at bad led obviously...
I can't put the board near LED 1 and I don't have the possibility to use repeaters...
There is a way to shift start led compared to where is the signal cable connected at program?
Thanks

toblum, 2019-04-21:
Hi @Maurotb,

that sounds like a faulty LED somewhere in the first 100. Connecting the strips somwhere in the middle isn't supported because of the way these strips work. Maybe you can find out which LED has the error and just get rid of this one.

Regards
Tobias

Maurotb, 2019-04-21:
I have not explained myself, everything works correctly. My strips are installed as an O.
To keep the length of the signal cable short, I need to connect it to a specific point, but so some effects where you need to have the start / end pixel in a specific position are bad to see. 
I should be able to move this starting point via the 2812fx segment function. I should perhaps send photos to make it clear

debsahu, 2019-05-01:
Please add changes to here:
https://github.com/toblum/McLighting/blob/090bddd32b2705f5edb7ec999cb26c50dc470151/Arduino/McLighting/McLighting.ino#L260

Maurotb, 2019-06-03:
i have created a pull request

closed 2019-06-03 by Maurotb