toblum/McLighting #391: WiFi disconnected, reconnecting!

bennyboom38, 2019-04-22:
Hello,

Once I have verify and upload the sketch , i get my strip leds in blue with AP mode, I enter the wifi credentials, then ESP connect on wifi, and leds strip turn on red but soon after this I get the following log and ESP disconnect from wifi : 

Should save config
Saving config: {"mqtt_host":"192.168.XX.XX","mqtt_port":"XXXX","mqtt_user":"","mqtt_pass":""}WiFiManager config FS Save success!
connected...yeey :)
MQTT active: 192.168.XX.XX:XXXX
Open http://192.168.XX.XX/ to open McLighting.
Use http://MyEsp.local/ when you have Bonjour installed.
New users: Open http://192.168.XX.XX/upload to upload the webpages first.

Couldn't find "/stripstate.json"
 Failure!
*WM: freeing allocated params!
MQTT disconnected, reconnecting!
Attempting MQTT connection 1 / 4 ...
MQTT connected!
MQTT topic in: MyEsp/in
MQTT topic out: MyEsp/out
SET_MODE: 0 0
Saving Strip cfg: {"pixel_count":24,"rgb_order":82,"pin":5}
MQTT: Send [home/MyEsp_ha/state/out]: {"state":"OFF","color":{"r":255,"g":0,"b":0},"brightness":196,"color_temp":327,"speed":196,"effect":"Static"}
Heap size: 21136
WiFi disconnected, reconnecting!
WiFi disconnected, reconnecting!
WiFi disconnected, reconnecting!
WiFi disconnected, reconnecting!
WiFi disconnected, reconnecting!
WiFi disconnected, reconnecting!
WiFi disconnected, reconnecting!
WiFi disconnected, reconnecting!

Could you help me ? 

Regards

Ben

bennyboom38, 2019-04-22:
I have disable MQTT and HomeAssistant and everything is working well.
Do you have an idea why MQTT and HomeAssistant could stop wifi working well ? 

Thanks a lot

debsahu, 2019-05-01:
That is weird, the code should be able to connect to MQTT and integrate with HA seamlessly.

HiranChaudhuri, 2019-05-06:
Did you enable MQTT or AMQTT? I think I had such problems in the beginning and stabilized like you by not using many options. Then I introduced AMQTT and it works as needed.

bennyboom38, 2019-05-07:
> 
> 
> That is weird, the code should be able to connect to MQTT and integrate with HA seamlessly.

Hello debsahu,

I have just try with another Wemos D1 Mini board  and it seems to work with Home Assistant & MQTT .

My definitions.h : 

#define ENABLE_MQTT  
#define ENABLE_HOMEASSISTANT

My configuration.yaml  (home assistant config file)

light:
  - platform: mqtt
    schema: json
    name: "<my device name>"
    state_topic: "homeassistant/<my device>/state/out"
    command_topic: "homeassistant/<my device>/state/in"
    effect: true
    effect_list:
    ######
      - "Static"
      - "Blink"
      - "Breath"
      - "Color Wipe"
      - "Color Wipe Inverse"
      - "Color Wipe Reverse"
      - "Color Wipe Reverse Inverse"
      - "Color Wipe Random"
      - "Random Color"
      - "Single Dynamic"
      - "Multi Dynamic"
      - "Rainbow"
      - "Rainbow Cycle"
      - "Scan"
      - "Dual Scan"
      - "Fade"
      - "Theater Chase"
      - "Theater Chase Rainbow"
      - "Running Lights"
      - "Twinkle"
      - "Twinkle Random"
      - "Twinkle Fade"
      - "Twinkle Fade Random"
      - "Sparkle"
      - "Flash Sparkle"
      - "Hyper Sparkle"
      - "Strobe"
      - "Strobe Rainbow"
      - "Multi Strobe"
      - "Blink Rainbow"
      - "Chase White"
      - "Chase Color"
      - "Chase Random"
      - "Chase Rainbow"
      - "Chase Flash"
      - "Chase Flash Random"
      - "Chase Rainbow White"
      - "Chase Blackout"
      - "Chase Blackout Rainbow"
      - "Color Sweep Random"
      - "Running Color"
      - "Running Red Blue"
      - "Running Random"
      - "Larson Scanner"
      - "Comet"
      - "Fireworks"
      - "Fireworks Random"
      - "Merry Christmas"
      - "Fire Flicker"
      - "Fire Flicker (soft)"
      - "Fire Flicker (intense)"
      - "Circus Combustus"
      - "Halloween"
      - "Bicolor Chase"
      - "Tricolor Chase"
      - "ICU"
      - "E131"
    brightness: true
    color_temp: true
    rgb: true
    optimistic: false
    qos: 0
    retain: true

input_number:
  neopixel_animation_speed:
    name: NeoPixel Animation Speed
    initial: 200
    min: 0
    max: 255
    step: 5

Problem is that home assistant see the light off instead of on, so the switch on home assitant does not work.
But when I switch off the led by MCLightning frontend, I can switch on with home assitant , but status is still wrong.

I don't have the panel to change color or effect, only the brightness setting is working, see the screenshot: 

[image]

bennyboom38, 2019-05-07:
> 
> 
> Did you enable MQTT or AMQTT? I think I had such problems in the beginning and stabilized like you by not using many options. Then I introduced AMQTT and it works as needed.

I have tried MQTT, and now MQTT seems working but with not all settings, I try to configure to use AMQTT to let home assistant configure by discovering

HiranChaudhuri, 2019-05-07:
I believe your HomeAssistant configuration needs improvement. However I only configured HomeAssistant to access MQTT and perform AutoDiscovery.
Did you enable HomeAssistant support (ENABLE_HOMEASSISTANT) in definitions.h?
For me McLightning sends a message to a specific topic offering itself to HomeAssistant. After that HomeAssistant knows about the device and it's capabilities, and my settings dialog contains a lot more than your screenshot shows...

closed 2019-07-05 by toblum