toblum/McLighting #392: Ha autodiscover not work :(

Maurotb, 2019-04-29:
Hi, 
i have followed this how to
https://github.com/toblum/McLighting/wiki/Homeassistant-integration
to enable autodiscover of mqtt mclighting but now work...
Mclighting  send only this
1556551984: Sending PINGRESP to McLighting-Ingresso
1556551995: Received PINGREQ from McLighting-Ingresso
1556552011: Received PUBLISH from McLighting-Ingresso (d0, q0, r0, m0, 'McLighting-Ingresso/out', ... (37 bytes))
1556552011: Received SUBSCRIBE from McLighting-Ingresso
1556552011:     McLighting-Ingresso/in (QoS 0)
1556552011: Sending SUBACK to McLighting-Ingresso
1556552011: Received PUBLISH from McLighting-Ingresso (d0, q0, r0, m0, 'McLighting-Ingresso/status', ... (6 bytes))
1556552011: Sending PUBLISH to McLighting-Ingresso (d0, q0, r1, m0, 'home/McLighting-Ingresso_ha/state/in', ... (16 bytes))

Not send the PUBLISH /config and HA not discover it.
If i set manually in configuration.yaml it work, nut, how enable autodiscover?
HA 0.89.1
McLighting latest
Thanks

debsahu, 2019-05-01:
Follow instructions from here: https://github.com/toblum/McLighting/wiki/Homeassistant-integration#homeassistant-mqtt-auto-discovery

Maurotb, 2019-05-01:
I have followed instruction, only difference is i use platform.io...

debsahu, 2019-05-01:
PIO works great, shouldnt matter.

1. Is MQTT discovery enabled in HA https://www.home-assistant.io/docs/mqtt/discovery/ ?
2. What is the output of `http://<ESP_IP_ADDRESS>/esp_status`?

Maurotb, 2019-05-01:
Yes, mqtt discovery work in my HA for other device connected with mosquitto and zigbee2mqtt

This is my HA config
mqtt:
  discovery: true
  discovery_prefix: homeassistant
  broker: 192.168.210.10
  port: 1883
  client_id: home-assistant
  username: xxx
  password: xxx

First McLighting

{"HOSTNAME":"McLighting-Ingresso","version":"2.2.4","heap":12152,"sketch_size":658768,"free_sketch_space":1437696,"flash_chip_size":16777216,"flash_chip_real_size":16777216,"flash_chip_speed":40000000,"sdk_version":"3.0.0-dev(c0f7b44)","core_version":"2_5_0","cpu_freq":80,"chip_id":1589487,"animation_lib":"WS2812FX","pin":5,"number_leds":896,"button_mode":"ON","amqtt":"ON","home_assistant":"ON","legacy_animations":"ON","esp8266_http_updateserver":"ON","state_save":"SPIFFS"}

Second

{"HOSTNAME":"McLighting-Corridoio","version":"2.2.4","heap":11296,"sketch_size":658784,"free_sketch_space":1437696,"flash_chip_size":16777216,"flash_chip_real_size":16777216,"flash_chip_speed":40000000,"sdk_version":"3.0.0-dev(c0f7b44)","core_version":"2_5_0","cpu_freq":80,"chip_id":1589487,"animation_lib":"WS2812FX","pin":5,"number_leds":975,"button_mode":"ON","amqtt":"ON","home_assistant":"ON","legacy_animations":"ON","esp8266_http_updateserver":"ON","state_save":"SPIFFS"}

debsahu, 2019-05-01:
Did you uncomment `#define MQTT_HOME_ASSISTANT_SUPPORT` in **definitions.h**?
https://github.com/toblum/McLighting/blob/090bddd32b2705f5edb7ec999cb26c50dc470151/Arduino/McLighting/definitions.h#L20

Maurotb, 2019-05-01:
Harg! No :( 
Sorry for my error

debsahu, 2019-05-01:
That was the first instruction on https://github.com/toblum/McLighting/wiki/Homeassistant-integration#homeassistant-mqtt-auto-discovery !

Please reopen if you are still facing issues.

closed 2019-05-01 by debsahu