toblum/McLighting #395: HomeAssistant Auto-Discovery does not work

HiranChaudhuri, 2019-05-04:
I compiled McLighting as shown in the documentation however using latest library versions. Then one by one I turned on additional features, also installed an MQTT broker and configured both McLighing and HomeAssistant to access it. After some time I was able to see both clients connect to the broker. But not yet would they talk to each other.

So I turned on Auto-Discovery, but it seems the message advertizing McLighting never pops up in my broker (checked with the broker's admin UI).

Detailed protocol of what I had done is in the PDF. What may have gone wrong?

[My McLightning Experience.pdf](https://github.com/toblum/McLighting/files/3144885/My.McLightning.Experience.pdf)

debsahu, 2019-05-04:
Please provide contents of **definitions.h** and confirm the version of MQTT you are using?

HiranChaudhuri, 2019-05-05:
I am sorry to see the comment I throught to have posted into this thread did not make it online.

Actually I found that HomeAssistant must have received the message immediately as there were some hints in the logs. So as soon as I turned off my custom view HomeAssistant would display all orphaned devices (devices not included in other groups), and McLightning was there. :-)

I still do not know why the Artemis UI did not show that queue. Here is the content of my definitions.h:

`//#define USE_WS2812FX_DMA       // LED_PIN is ignored & set to RX/GPIO3  Uses WS2812FX, see: https://github.com/kitesurfer1404/WS2812FX
//#define USE_WS2812FX_UART1     // LED_PIN is ignored & set to D4/GPIO2  Uses WS2812FX, see: https://github.com/kitesurfer1404/WS2812FX
//#define USE_WS2812FX_UART2     // LED_PIN is ignored & set to TX/GPIO1  Uses WS2812FX, see: https://github.com/kitesurfer1404/WS2812FX

// Neopixel
#define LED_PIN 14       // LED_PIN (14 / D5) where neopixel / WS2811 strip is attached
#define NUMLEDS 24       // Number of leds in the strip
//#define LED_TYPE_WS2811  // Uncomment if LED type uses 400 KHz (classic 'v1' (not v2) FLORA pixels, WS2811 drivers)
#define BUILTIN_LED 2    // ESP-12F has the built in LED on GPIO2, see https://github.com/esp8266/Arduino/issues/2192
#define BUTTON 4         // Input pin (4 / D2) for switching the LED strip on / off, connect this PIN to ground to trigger button.

#define HOSTNAME "McLighting01"   // Friedly hostname

#define HTTP_OTA             // If defined, enable ESP8266HTTPUpdateServer OTA code.
//#define ENABLE_OTA         // If defined, enable Arduino OTA code.
#define ENABLE_AMQTT         // If defined, enable Async MQTT code, see: https://github.com/marvinroger/async-mqtt-client
//#define ENABLE_MQTT        // If defined, enable MQTT client code, see: https://github.com/toblum/McLighting/wiki/MQTT-API
#define ENABLE_HOMEASSISTANT // If defined, enable Homeassistant integration, ENABLE_MQTT or ENABLE_AMQTT must be active
#define ENABLE_BUTTON        // If defined, enable button handling code, see: https://github.com/toblum/McLighting/wiki/Button-control
#define MQTT_HOME_ASSISTANT_SUPPORT // If defined, use AMQTT and select Tools -> IwIP Variant -> Higher Bandwidth
#define ENABLE_LEGACY_ANIMATIONS // Dont disbale this for now
//#define ENABLE_E131              // E1.31 implementation
#define USE_HTML_MIN_GZ             //comment for using index.htm & edit.htm from SPIFFs instead of PROGMEM
//#define CUSTOM_WS2812FX_ANIMATIONS  //uncomment and put animations in "custom_ws2812fx_animations.h" 

#ifdef ENABLE_E131
  #define START_UNIVERSE 1                    // First DMX Universe to listen for
  #define END_UNIVERSE 2                      // Last Universe to listen for, starting at UNIVERSE
                                              // MUST: END_UNIVERSE >= START_UNIVERSE
#endif

//#define WIFIMGR_PORTAL_TIMEOUT 180
//#define WIFIMGR_SET_MANUAL_IP

#ifdef WIFIMGR_SET_MANUAL_IP
  uint8_t _ip[4] = {192,168,0,128};
  uint8_t _gw[4] = {192,168,0,1};
  uint8_t _sn[4] = {255,255,255,0};
#endif

#ifdef MQTT_HOME_ASSISTANT_SUPPORT
//  #define MQTT_HOME_ASSISTANT_0_87_SUPPORT // Comment if using HA version < 0.87
#endif

#if defined(USE_WS2812FX_DMA) and defined(USE_WS2812FX_UART)
#error "Cant have both DMA and UART method."
#endif
#if defined(ENABLE_MQTT) and defined(ENABLE_AMQTT)
#error "Cant have both PubSubClient and AsyncMQTT enabled. Choose either one."
#endif
#if ( (defined(ENABLE_HOMEASSISTANT) and !defined(ENABLE_MQTT)) and (defined(ENABLE_HOMEASSISTANT) and !defined(ENABLE_AMQTT)) )
#error "To use HA, you have to either enable PubCubClient or AsyncMQTT"
#endif
#if ( !defined(ENABLE_HOMEASSISTANT) and defined(MQTT_HOME_ASSISTANT_SUPPORT) )
#error "To use HA support, you have to either enable Homeassistant component"
#endif

// parameters for automatically cycling favorite patterns
uint32_t autoParams[][4] = {  // color, speed, mode, duration (milliseconds)
  {0xff0000, 200,  1,  5000}, // blink red for 5 seconds
  {0x00ff00, 200,  3, 10000}, // wipe green for 10 seconds
  {0x0000ff, 200, 14,  5000}, // dual scan blue for 5 seconds
  {0x0000ff, 200, 45, 15000}  // fireworks for 15 seconds
};
[... 88 lines cut ...]
#ifdef ENABLE_STATE_SAVE_EEPROM
  char current_state[32];               // Keeps the current state representation
  char last_state[32];                  // Save the last state as string representation
  unsigned long time_statechange = 0;   // Time when the state last changed
  int timeout_statechange_save = 5000;  // Timeout in ms to wait before state is saved
  bool state_save_requested = false;    // State has to be saved after timeout
#endif
#ifdef ENABLE_STATE_SAVE_SPIFFS
  bool updateStateFS = false;
#endif

// Button handling
#ifdef ENABLE_BUTTON
  #define BTN_MODE_SHORT "STA| 1|  0|245|196|255|255|255"   // Static white
  #define BTN_MODE_MEDIUM "STA| 1| 48|245|196|255|102|  0"    // Fire flicker
  #define BTN_MODE_LONG "STA| 1| 46|253|196|255|102|  0"  // Fireworks random

  unsigned long keyPrevMillis = 0;
  const unsigned long keySampleIntervalMs = 25;
  byte longKeyPressCountMax = 80;       // 80 * 25 = 2000 ms
  byte mediumKeyPressCountMin = 20;     // 20 * 25 = 500 ms
  byte KeyPressCount = 0;
  byte prevKeyState = HIGH;             // button is active low
  boolean buttonState = false;
#endif`

HiranChaudhuri, 2019-05-07:
As AutoDiscovery works for me now even after renaming the device (it's hostname and queue names are all customized with chip id) I suggest we can close this issue.

closed 2019-05-07 by HiranChaudhuri