toblum/McLighting #396: How to change MQTT server address without reprogramming the ESP

roesl, 2019-05-05:
I installed a new MQTT Server so I have to change the server address.
Is there a tool to do that ?

debsahu, 2019-05-06:
Use `http://<ip_address>/reset_wlan`

HiranChaudhuri, 2019-05-06:
I did not like the idea of resetting the WLAN so my only chance would be to use my mobile to enter all the critical data, so I extended McLighting.ino by these lines (starting from line 692, if that is still applicable):
'
  //process request to see what config changes are there, then print mqtt settings
  server.on("/mqtt_status", HTTP_GET, []() {
    bool configDirty = false;
  
    if(server.hasArg("mqtt_host")) {
      server.arg("mqtt_host").toCharArray(mqtt_host, 64);
      configDirty = true;
    }
    if(server.hasArg("mqtt_port")) {
      server.arg("mqtt_port").toCharArray(mqtt_port, 6);
      configDirty = true;
    }
    if(server.hasArg("mqtt_user")) {
      server.arg("mqtt_user").toCharArray(mqtt_user, 32);
      configDirty = true;
    }
    if(server.hasArg("mqtt_pass")) {
      server.arg("mqtt_pass").toCharArray(mqtt_pass, 32);
      configDirty = true;
    }

    DynamicJsonDocument jsonBuffer(JSON_OBJECT_SIZE(21) + 1500);
    JsonObject json = jsonBuffer.to<JsonObject>();

    if(configDirty) {
      (writeConfigFS(configDirty)) ? json["message"] = "WiFiManager config FS Save success! Restart recommended.": json["message"] = "WiFiManager config FS Save failure!";
    }
    
    json["mqtt_host"] = mqtt_host;
    json["mqtt_port"] = mqtt_port;
    json["mqtt_user"] = mqtt_user;
    json["mqtt_pass"] = mqtt_pass;
    
    String json_str;
    serializeJson(json, json_str);
    server.sendHeader("Access-Control-Allow-Origin", "*");
    server.send(200, "application/json", json_str);
  });

'

debsahu, 2019-05-10:
Thanks @HiranChaudhuri !

closed 2019-05-10 by debsahu