toblum/McLighting #397: Problems with future HomeAssistant versions

HiranChaudhuri, 2019-05-08:
I use HomeAssistant 0.92.2 and keep getting warnings about deprecated configuration directives on_command_type and platform=mqtt_json", which would be a breaking change (change that breaks functionality) soon.

The values are not part of my HomeAssistant configuration. Instead, they are injected via AutoDiscovery from McLighting itself. To change this, I would have to change the McLighting code advertizing itself to HomeAssistant.

Diving into the code I saw that all the necessary changes are actually done in #ifdef / #ifndef directives depending on MQTT_HOME_ASSISTANT_0_87_SUPPORT. So I was wondering why I had not set  the symbol.

It seems the comment in that line is wrong. It should read as
` #define MQTT_HOME_ASSISTANT_0_87_SUPPORT // Comment if using HA version > 0.87 `

This is an extremely easy fix as everything else is already done.

debsahu, 2019-05-10:
Sorry I dont understand.

I see that you are using HA version 0.92.2 which is higher than 0.87, so in your case you will need to only uncomment https://github.com/toblum/McLighting/blob/090bddd32b2705f5edb7ec999cb26c50dc470151/Arduino/McLighting/definitions.h#L20
And dont touch https://github.com/toblum/McLighting/blob/090bddd32b2705f5edb7ec999cb26c50dc470151/Arduino/McLighting/definitions.h#L42

If that is the case, only `json["schema"] = "json";` is sent which will not give you the warning you get in HA
https://github.com/toblum/McLighting/blob/090bddd32b2705f5edb7ec999cb26c50dc470151/Arduino/McLighting/request_handlers.h#L1021

HiranChaudhuri, 2019-05-12:
I just verified, and your statement is absolutely true. I may have gotten confused with comment/uncomment when starting off on McLighting.

With current versions of HomeAssistant the symbol needs to be defined, and it is defined per default.
Maybe the documentation (the comment in definitions.h) could be a bit more verbose, otherwise it's fine.

closed 2019-05-12 by HiranChaudhuri