toblum/McLighting #398: MQTT Publish to HA slow

Maurotb, 2019-05-10:
Hi,
i have mosquitto as broker and HA ,
if i switch on/off my led in McLighting web gui, in mosquitto log i see imediatly
1557498027: Received PUBLISH from McLighting-Ingresso (d0, q0, r0, m0, 'McLighting-Ingresso/out', ... (6 bytes))

and after 1 - 2 seconds 

1557498032: Received PUBLISH from McLighting-Ingresso (d0, q1, r1, m17, 'home/McLighting-Ingresso_ha/state/out', ... (110 bytes))
1557498032: Sending PUBACK to McLighting-Ingresso (Mid: 17)
1557498032: Sending PUBLISH to home-assistant (d0, q0, r0, m0, 'home/McLighting-Ingresso_ha/state/out', ... (110 bytes))

In my HA i have a delay of 1-2 seconds in refresh status of my Led if i switch from McLighting gui.
This is a problem for me, i need some 'realtime' automation....
Thanks

debsahu, 2019-05-10:
The 5s delay is intentional, so that ESP8266 doesn't reset every time someone spams the controls on HA. If you want what you say you need, please change all occurances of `ha_send_data.once(5, tickerSendState);` to `ha_send_data.once(0, tickerSendState);` in **request_handlers.h**, which I highly not recommend doing!

closed 2019-05-10 by debsahu

M4lik, 2019-05-12:
Although i would say that 5 seconds is quite high, as this is more than HA waits before it resets the status of the entity. I would suggest to lower it a bit. Maybe to 2-3s?