toblum/McLighting #400: 60 NeoPixels is too much for my ESP?

AtiShotgunFox, 2019-05-18:
After turning on the ESP everything was fine but after setting up Wi-Fi connection and turning on NeoPix strip program started lagging: Tearing animations and some white flashing pixels. When I changed static color I had to wait at least 5 seconds till the color was set on the strip.

toblum, 2019-05-18:
Hi @AtiShotgunFox,

60 neopixels are definitely no problem. There are setups with a lot more working flawlessly. 
Flashing pixels could be a power problem. This setup needs a stable power supply that is beefy enough.
How does your definitions.h look like? Did you do any changes?

Regards
Tobias

HiranChaudhuri, 2019-05-21:
I am running already 119 WS2812 pixels. during test I went to 240 (before I cut the strip).
Right now I am testing up to 240 SK6812. When satisfied I'll also cut this strip but so far I do not see effects you describe.

The biggest lag I saw was using MQTT, but it was not 5 seconds. How do you control the unit?

closed 2019-07-05 by toblum