toblum/McLighting #402: Scripting in HomeAssistant

HiranChaudhuri, 2019-05-21:
As I am trying to control McLighting from HomeAssistant I have great success via the HomeAssistant UI - but then HomeAssistant is about automation, right?

Thank you for providing the software and even the documentation up to this point:
https://github.com/toblum/McLighting/wiki/Homeassistant-integration

Would it be possible to extend it a little how to create scripts that control McLighting? My scripts either generate error messages or seem to have no effect at all.

deppen12, 2019-05-24:
pla make some video, how you did this. i'm new to this. i can control lights from IP base web page not online .

HiranChaudhuri, 2019-05-24:
> pla make some video, how you did this. i'm new to this. i can control lights from IP base web page not online .

Please do not hijack this thread which is about scripting and open a separate issue. I'm willing to help even though I am not the video creation type of guy.

deppen12, 2019-05-25:
is this call hijacking? i didnt knew. i was asking you for help. can you share your files or setting how you did this

M4lik, 2019-05-27:
@deppen12
If you want to use McLighting with Homeassitant please refer to the wiki section [Homeassistant integration](https://github.com/toblum/mclighting/wiki/Homeassistant-integration).

@HiranChaudhuri
I'm sorry that I can't help you atm. But as I am also playing around with HA I will report anything I find out in the next days.