toblum/McLighting #403: Feature Request: fading on color change in Static mode

sladaar, 2019-05-23:
First of all, many many thanks Tobias for this, i searched a lot for a valid code and i've been using yours for a year by now. So thanks again!

I was wondering if it's possibile to edit/add a fading between colors in static mode, because i doubt a little it could be hardware related. I'm no programmer but feel free to point me where to edit the code, I'll try my best.

I use mqtt to control the board, via openHab. (pin 12, 300 led ws2811)

Many thanks in advice!

debsahu, 2019-05-23:
You can use "rainbow" effect and slow the speed to your liking? Or are you referring to fading when you select a new color? All animations are dependent on WS2812FX, and I would ask you to refer to https://github.com/kitesurfer1404/WS2812FX/issues/124 to get some idea if latter is your intention.

sladaar, 2019-05-24:
> 
> 
> You can use "rainbow" effect and slow the speed to your liking? Or are you referring to fading when you select a new color? All animations are dependent on WS2812FX, and I would ask you to refer to [kitesurfer1404/WS2812FX#124](https://github.com/kitesurfer1404/WS2812FX/issues/124) to get some idea if latter is your intention.

Thanks for the reply!
Yes i was referring to the static mode, so it would be great to see a fading between the old color and the new selected one ( i mean, like commercial products).
Many thanks for the link, i'll check it!

To be honest, i thought at this feature because the fastest way i found to control the nodemcu+mclighting with Google Assistant was by "cloning" (in openHab) the commands received by a Yeelight lamp, and in testing phase i spotted the difference (the yeelight fades, mclighting not).

I'll post my result, if positive!

DarkFox, 2019-07-22:
+1 for this feature, and in addition, support for the "transition" feature from Home Assistant, which lets you set how many seconds the transition should take. Currently, if you define a transition time from HA, the lights don't react at all.