toblum/McLighting #404: Unable to set LED > 99 via MQTT

Hypfer, 2019-05-27:
Sending `+99d2ff` works, however `+100d2ff` does not.

toblum, 2019-05-27:
Hi @Hypfer,

that's correct. You need to user hexadecimal values. Each position two digits.
After 99 comes 9A, so to set LED 100 to white you would use 9AFFFFFF.

Regards
Tobias

Hypfer, 2019-05-27:
@toblum Nope. Doesn't work. `+9AFFFFFF` turns on the 10th LED.

toblum, 2019-05-27:
What value did you set for the number of LEDs? What is returned when you call http://mc_lighting_ip/esp_status

Hypfer, 2019-05-27:
```
"HOSTNAME": "McLighting01",
"version": "2.2.4",
"heap": 17888,
"sketch_size": 691200,
"free_sketch_space": 2453504,
"flash_chip_size": 4194304,
"flash_chip_real_size": 4194304,
"flash_chip_speed": 40000000,
"sdk_version": "3.0.0-dev(c0f7b44)",
"core_version": "2_5_0",
"cpu_freq": 80,
"chip_id": 1458415,
"animation_lib": "WS2812FX",
"pin": 5,
"number_leds": 239,
"button_mode": "ON",
"amqtt": "ON",
"home_assistant": "ON",
"legacy_animations": "ON",
"esp8266_http_updateserver": "ON",
"state_save": "SPIFFS"
}
```

239 LEDs is (sadly) correct. I've cut the wrong amount of LEDs :(

toblum, 2019-05-27:
Oh, sorry, I was wrong. With this command it's indeed only possible to adress only 99 LEDs, because the number is handles as a integer here: https://github.com/toblum/McLighting/blob/090bddd32b2705f5edb7ec999cb26c50dc470151/Arduino/McLighting/request_handlers.h#L141

Here you can try to change the line to: 
templed[2] = 0x00;
uint8_t led = strtol(templed, NULL, 16);

To be able to use hex values.

Regards
Tobias

HiranChaudhuri, 2019-05-28:
> What value did you set for the number of LEDs? What is returned when you call http://mc_lighting_ip/esp_status

Just a comment here: I noticed that this query returns the compiled in number of LEDs, not the actual value which can be changed at runtime.
https://github.com/toblum/McLighting/blob/master/Arduino/McLighting/McLighting.ino#L631

Try your request after accessing http://mc_lighting_ip/pixelconf?ct=1423

zen85, 2019-11-20:
> Oh, sorry, I was wrong. With this command it's indeed only possible to adress only 99 LEDs, because the number is handles as a integer here: [...]

that works. so now i can go as high as 255. i guess i can not go higher than that. not a terribly pressing problem but it would still be nice to go higher than that. i have about 1000 leds in a row in my setup.