toblum/McLighting #405: Battery power

iansexton, 2019-05-30:
Hi, I really like McLighting! Great work guys :)

I have a project which is powered from a LiPo battery. I intend to monitor the battery voltage using the ESP8266 ADC - if it drops too low everything will shut down, LEDs off & ESP8266 into deep sleep.
I'm wondering what is the most elegant way to do this?
My initial thoughts are to use a 'Ticker' to periodically check the voltage but I thought I'd ask if there was a better way.
Maybe you could add a 'user ticker' utility? I can imagine a lot of cases where this might be useful, eg read a temperature probe & change the display colour/pattern. I imagine it could be difficult to provide something that would please everyone, however my immediate need is simple.
Ian

toblum, 2019-05-31:
Hi @iansexton,
using a ticker or just checking millis() in the main loop to check is a time interval has elapsed is also the first thing that comes into my mind.
As the Ticker library is already in place, why don't you go with it? 
Doing something more generic is not easy as it would need custom code anyway.

Regards
Tobias

iansexton, 2019-06-04:
@toblum 
Thx, we're thinking the same way - Your coding is great, easy to slip in my mods.
Cheers
Ian

iansexton, 2019-06-06:
Hi @Tobias
Glad you agree :) I have a habit of doing things the hard way first &
realising afterwards...

Cheers
Ian