toblum/McLighting #408: Feature Request: change luminosity on off

Maurotb, 2019-06-01:
For usability o mclighing in my home, is possible to make
from off to on (whatever effect)-> luminosity from 0 to 254 gradually
from on to off (whatever effect)-> luminosity from 254 to 0 gradually
Thanks

toblum, 2019-06-04:
Hi @Maurotb,

do you think of some fade effect to fade between two colors?

Regards
Tobias

Maurotb, 2019-06-04:
in some effect...
for example, in my house i love to use running light and static (i use led strip for illumination,not only for effect) , but is only on or off, some fade in fade out when power on / off led is beautiful (hue led make this).
I'm coding this for my use in running light, but i'm not a good programmer and i see is very very 
hard to make this fade for all effect, but in real house illumination, this is a good feature...
I have try changing luminosity, but it flicker... now i'm tring to make a custom running light effect with a big change in loop function, i't work but is an hard hack...

toblum, 2019-06-17:
Yes, this seems not to be easy to accomplish. I may put it on the list for future enhancements.

Maurotb, 2019-06-18:
Good.

If it can be useful, this is my approach (along with my pull request # 407 to be able to move the LED 0,
Thanks to @moose4lord)

//1) CustomShow function to move led 0 and to fadein/fadeout during powe on/power off
void CustomShow(void) {
  static uint8_t tmpPixels[NUMLEDS * 4]; // allocate space for a copy of the ADafruit pixel data
  uint8_t bytesPerPixel = strip->getNumBytes() / strip->getLength(); // 3=RGB, 4=RGBW
  // copy the Adafruit pixel array
  memcpy(tmpPixels, strip->getPixels(), strip->getNumBytes());

  //MOVE LED 0 -> offset the data in the Adafruit pixel array
  memcpy(strip->getPixels() + OFFSET * bytesPerPixel, tmpPixels, (NUMLEDS - OFFSET) * bytesPerPixel);
  memcpy(strip->getPixels(), tmpPixels + (NUMLEDS - OFFSET) * bytesPerPixel, OFFSET * bytesPerPixel);
  uint8_t *pixels = strip->getPixels(); // the raw Adafruit pixel array

  // Adjust myBrightness up or down depending on myBrightnessTarget
  if(brightness_fade == 5 && brightness_target == 0){
    brightness_fade = 0;
    if(strip->isRunning()) strip->stop(); //should clear memory
  }
  if(brightness_fade != brightness_target){
    if(brightness_fade < brightness_target) {
      brightness_fade=brightness_fade+5;
    }else{
      brightness_fade=brightness_fade-5;
    }

  }
  }
  // adjust the data in the Adafruit pixel array by myBrightness
  for(uint16_t i=0; i <strip->getNumBytes(); i++) {
    pixels[i] = (brightness_fade * pixels[i]) / 256;
  }

  // run the standard Adafruit show() function
  strip->Adafruit_NeoPixel::show();
  // copy the original pixel data back to the Adafruit pixel array
  memcpy(strip->getPixels(), tmpPixels, strip->getNumBytes());
}

And some hard modification in loop function which are to be completely reviewed since I had to be very messy...

//2)This function is to blend colors in static effect from one color to another (like the philips hue)
void blend() {
  if(millis() > blendTimer + BLEND_RATE) { // time to update color?
    uint32_t newColor = strip->color_blend(blend_start_color, blend_end_color, blendAmount); // calc the new color
    strip->setColor(newColor); // set the new color
    blendTimer = millis();       // reset the timer
    blendAmount++;               // incr the blend amount
  }
}

uint32_t Color(byte r, byte g, byte b)
{
  uint32_t c;
  c = r;
  c <<= 8;
  c |= g;
  c <<= 8;
  c |= b;
  return c;
}

Another wery useful function which I implemented, is a delay in activating the strip. 
In fact I always remove the power supply to the strip, 
and from the home assistant via mqtt if check strip has been turned on, 
then i activate the power supply with a shelly switch.
In this situation, the strip flashes quickly before it starts working properly.

I solved by delaying SET_MODE but i have to inform HA hat the strip is on even if it is not immediately, so HA turns on the shelly.

if (mode == SET_MODE) {
  if (!strip->isRunning()) {
    if(!ha_send_data.active())  ha_send_data.once(5, tickerSendState);
    delay(DELAY_START);
  }    
   DBG_OUTPUT_PORT.printf("SET_MODE: %d %d\n", ws2812fx_mode, mode);
   if(strip) strip->setMode(ws2812fx_mode);
   if(strip) strip->trigger();
   prevmode = SET_MODE;
   mode = SETCOLOR;
}

toblum, 2019-07-05:
Thank for your feedback @Maurotb

closed 2019-07-05 by toblum